imbushuo/mac-precision-touchpad #18: "The folder you specified doesn't contain a compatible software driver for your device"

mihe, 2017-09-11:
I'm trying to get your recently released [Milestone 3](https://github.com/imbushuo/mac-precision-touchpad/releases/tag/mt2-m3) drivers to work with my Magic Trackpad 2, and I feel like I'm failing on something basic, so excuse me for posting this under Issues.

Basically I get [this dialog](https://i.imgur.com/XsvwOAv.png) whenever I point my (what Windows detects as) "HID-compliant mouse" to use your INF file. I double-checked that I picked the right platform (x64), and that I didn't pick the wrong mouse device, so I know it's not that. Any idea what the issue might be?

My Windows version is Windows 10 Pro, version 1703, build 15063.540.

By the way, I'm excited to see a project like this and I'd be glad to help out in any way I can. I can't say I've done any driver development, but I've done plenty of C++ (and some C) in Visual Studio. I'd happily do testing as well. My email's in my profile.

imbushuo, 2017-09-11:
The mouse is a HID Top-Level Collection exposed by the actual parent USB device. Use "view by connection" options in Device Manager, you should see a "USB Input Device" as the parent device of the mouse. The inf file matches that "USB Input Device", not the mouse itself.

Alternatively, you can right-click on inf that matches your machine architecture (inf in inner directory with corresponding catalog file (.cat)) and click "Install", then you are good to go. Test signing is not required.

Remember there's issues existing in this driver(see some in Issues section), feel free if you hit additional issues. At least UMDF driver won't crash your PC....

imbushuo, 2017-09-11:
I am glad if you want to join. UMDF driver development is far more easy then KMDF.

However, there are few things to one: you need to install Windows Driver Kit with Visual Studio 2015 (C++ features must be installed). Installing WDK will break UWP development in VS2015(VS2017 is fine). Secondly, do not attempt to provision driver development test agent on your development machine or it will freak you out. Use an alternative device or virtual machine instead. 64bit preferred.

mihe, 2017-09-12:
Awesome, I got it to work now. I had to turn on Test Mode in Windows in order to get past the signed driver requirement, but apart from that it was as you described.

Some issues I noticed right away:
- The cursor locks up sometimes for reasons I can't figure out. I feel like it might be related to "Tap with a single finger to single-click", but I'm not sure.
- Scrolling with two fingers can take a couple of tries to actually get it to scroll sometimes.
- The three- and four-finger multi-tasking gestures seem very sensitive.

Not sure how much of this part of your drivers and how much is part of Windows Precision. Might just be that I need to play with the sensitivity setting in Windows as well. I'll create new issues if/when I can reproduce them accurately.

closed 2017-09-12 by mihe