imbushuo/mac-precision-touchpad #23: Missing installation instructions

cor, 2017-10-29:
Can someone please add installation instructions?

Nymeria815, 2017-11-05:
For me personally, it only works when plugged into the USB port. But basically you plug it in and it installs the windows drivers for it. Go to Device Manager and find the trackpad on there, click properties, change, update driver/Browse my computer/Let me pick/Have disk. Download the drivers off the main page, unextract, and then point the driver selection to C:\Users\YourUserName\Downloads\Drivers.1248(1)\Drivers\x64\Release\MagicTrackpad2PtpDevice

It'll be the .inf file. Then click apply and it'll say that the driver is unsigned, click install anyway. Then restart when it tells you to and that should be it. 

It might work wirelessly for everyone else though, so give that a shot first. Process would be the same.

imbushuo, 2017-11-06:
1) Download the zip package with binary if you don't want to compile it.
2) Go to `Drivers\x64\Release` directory
3) Install the `MagicTrackpad2PtpDevice.cer` certificate to your machine's root authority store.
4) Go to `MagicTrackpad2PtpDevice` directory.
5) Right click the inf file and install it. **It is not necessary to turn off driver signature enforcement and Secure Boot.**

It only works for Apple Magic Trackpad 2 w/ USB connection now; Windows's Bluetooth HID stack is totally tricky. Work for other MacBook devices is in progress, but I don't have a clear schedule yet - signal processing is tricky.

closed 2017-11-06 by imbushuo

cor, 2017-11-06:
Thanks a lot! Missing bluetooth support isn't an issue for me as I exclusively use my trackpad on a wired connection when using Windows :)

MilanBaca, 2017-11-07:
Could I ask,what functions this driver has?