imbushuo/mac-precision-touchpad #24: Cursor offten freeze for 2 - 3 s. Touchpad 2

MilanBaca, 2017-11-09:
Hello,

I just install a driver from zip archive. When I tap on icon for open application, the cursor will freeze on this icon for 2-3s sometime. Also sometimes it freeses when i  lift my finger from the surface of the touchpad and take it down for next moving with cursor.

imbushuo, 2017-11-09:
Looks like a tip switch detection issue (currently the pressure detection is not accurate enough).
Working on fixing that (altogether with multi-finger issues)

imbushuo, 2017-12-02:
Okay - a quick update from the hell of tons on quizzes and exams: I've re-implemented tip switch detection logic with Linux's `ABS_MT_TOUCH_MAJOR` and `ABS_MT_TOUCH_MINOR`. Multi-touch functionalities should be significantly improved then.

I will post related commits and binaries later.

sirburpalot, 2018-01-02:
I'm using the latest driver version (2014) with the Magic Trackpad 2, and the driver is pretty much unusable for me because two-finger gestures, as well as simply moving the pointer with one finger, only work ~85% of the time. Is anyone else experiencing this?

AGreatFish, 2018-01-02:
Yes, I am also experiencing this, @sirburpalot.
This driver seems to work perfectly with my Magic Trackpad 2, except for that fact that it has a lot of trouble detecting my fingers, especially when I only touch the trackpad with the tip of my finger.

PS: @imbushuo Thanks for developing this! Once your drivers have reached production quality, they will singlehandedly turn the Windows experience on a Mac from mediocre to amazing :)

imbushuo, 2018-01-02:
@AGreatFish @sirburpalot During the development of MacBook Pro devices, I noticed that I have misunderstood some values reported back from trackpad. MacBook Pro devices have better gesture experience as I fixed them for newer models. For Magic Trackpad 2, I will apply similar fixes in a few days.

BTW: Going back to US as my holiday approaches the end. Probably I will resume the project a week later...

imbushuo, 2018-01-09:
Will track with #37

closed 2018-01-09 by imbushuo

sirburpalot, 2018-07-14:
The latest update has brought major regressions, with usability on the Magic Trackpad 2 now being on par with some of the earliest releases. The cursor freezes constantly, and the only improvement is that pinch-to-zoom, which didn't work for some reason with the stable driver, now works again.