imbushuo/mac-precision-touchpad #33: MacBook Pro 2017

PaulWen, 2018-01-02:
Hi, thank you for your great work!!
I am really interested in a precision touchpad driver for the MacBook Pro 2017. Can you tell me if/when this model will be supported?

imbushuo, 2018-01-02:
Planned. New models use SPI-only trackpad, which needs additional research.

PaulWen, 2018-01-05:
Thank you @imbushuo. I just came across the following Git repo: [https://github.com/cb22/macbook12-spi-driver](url). Unfortunately, I am not able to evaluate if it helps regarding the SPI topic. But in case you have not seen it yet I wanted to point it out.

PaulWen, 2018-01-10:
If I understood Microsoft's documentation correctly an SPB Framework exists for developing SPI drivers. The SPB Framework is a KMDF extension library. It seems like a KMDF driver instead of a UMDF driver has to be written in order to support an SPI device.

> The SPB Framework Extension is a KMDF extension library. It handles the up-front processing of SPB request and the sequence in which they are handed to the controller driver. The SPB framework extension is designed to support I²C and SPI buses and may be appropriate for other buses with similar semantics. 
[(source)](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/simple-peripheral-bus--spb-)

An introduction to the SPB framework can be found [here](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/simple-peripheral-bus--spb-#spb-framework-extension).

tspitznas, 2018-02-22:
It will make my life infinitely better if or when you are able to get compatibility with MacBookPro13/14,3 SPI models. Keep up the fantastic work!

jourdant, 2018-06-11:
Hi @imbushuo really appreciate the work you are putting in - do you any updates on progress?

imbushuo, 2018-07-07:
See #ee.

closed 2018-07-07 by imbushuo