imbushuo/mac-precision-touchpad #37: Two-fingers gesture still hard to trigger

chaopeng, 2018-01-09:
I have tried latest release (Drivers.2014) with magic trackpad 2 on Windows 10 1607 14393.1944. Two-fingers gesture still hard to trigger. Sometime it recognize as three-fingers gesture.

imbushuo, 2018-01-09:
Working on this.

gastondassaro, 2018-01-17:
I love my trackpad, I use it on my MAC and it is indispensable, I am experiencing it on my PC since I live from the use of MS.Windows more than IOS, now that Ms.Windows supports gestures the trackpad is an indispensable tool for desktop computers . The work they are doing is impressive, I am sure that when it is a stable product they will have thousands of driver users around the world !!

imbushuo, 2018-01-19:
Implemented first set of improvement in https://github.com/imbushuo/mac-precision-touchpad/commit/eeefa15b004853ed958106c5a85f3854d76dd0df. Need to further adjust additional parameters and de-fuzzing algorithm for better experience.

MilanBaca, 2018-03-09:
Hi all, is there any progress with MT2?

bwet99, 2019-02-07:
very impressive work! the magic trackpad 2 is working on windows 10 , 1809, but, the two-finger right-click is *very difficult to trigger here. I can get it to do it occasionally. sometimes a very fast two finger click on a file icon will bring up the right click menu. for the taskbar, sometimes if I use index + ring finger together to click, works! very random behavior here. but, dbl click, two finger scroll in web page, and general mouse moves all working well.