imbushuo/mac-precision-touchpad #38: Trackpad frequently rejecting input (one finger/normal mousing)

Nymeria815, 2018-01-11:
(Sorry if this is too related to the two finger bug)

My Magic Trackpad 2 is performing inconsistently with normal mousing (one finger). It rejects at least 3 out of every five swipes I do. I've tried different angles/holding finger flat vs the tip but it behaves the same either way. I'm also getting the two finger bug with scrolling, but is anyone else having an issue moving the mouse normally? Wondering if it's on my end or not. Thanks!

imbushuo, 2018-01-12:
Issue specific to MT2. Track with #37

ChristopherHammond13, 2018-01-16:
Just thought I'd add in that I'm experiencing the same.
Windows 10 16299.192, MT2, Surface Book (MT2 connected via USB). I can try on my desktop too if you need reports?

Do let me know if I can help!

imbushuo, 2018-01-19:
Implemented first set of improvement in https://github.com/imbushuo/mac-precision-touchpad/commit/eeefa15b004853ed958106c5a85f3854d76dd0df. Need to further adjust additional parameters and de-fuzzing algorithm for better experience.

ChristopherHammond13, 2018-01-23:
As an update to above: I have found that the latest release (Drivers.2014) generally work okay except if my fingers are warm / clammy then the trackpad struggles with the input. No idea if this is helpful information or not but thought I'd add it for debugging. Thanks again for all your amazing work!

dinhngtu, 2018-01-23:
If you have a MT2, try the newest commit from the hid-v2 branch. Alternatively, try my fork (https://github.com/dinhngtu/mac-precision-touchpad/tree/mt2fix), which has some tweaks to increase the sensitivity.

Andersmidtbo, 2018-02-01:
I have the same problem, but it works more like once every 10th try :( But the multi touch features however works amazingly!

kakyoism, 2019-04-23:
Have the same here. Inconsistent single-finger and double-finger gestures. Will lose control every few seconds.

imbushuo, 2019-04-23:
This will mostly happen on Magic Trackpad 2 (though other models may experience it happen sometimes). DSP work is necessary, but currently, I have no plan on that.

Thanks,
Bingxing Wang

imbushuo, 2019-08-28:
Should be resolved with latest commit.