imbushuo/mac-precision-touchpad #39: MacBook Air 13" Mid-2013 (Not working)

Wonigox, 2018-01-12:
I installed the driver on my Macbook Air 13" (mid-2013 model) but it doesn't show up in the drivers list on my trackpad driver when updating it. Does the recent release support the macbook air 13" 2013 model or did I do something wrong in the installation?

I installed it by going to the x64\ReleaseSigned and then installing the certificate and going to x64\ReleaseSigned\AmtPtpDevice and installing the .inf file.

imbushuo, 2018-01-12:
I added MacBook Air support according to USB descriptor, I have not tested device bring-on, neither input experience (should have a similar issue of #35) with real machine now. 

Please report whether it works or not for you (closing issue directly confuses me).

tonynca, 2019-06-13:
I have the same issue. I tried manually installing this driver by replacing the stock Apple Multi-touch Device drivers with your own but the device fails to start.

Macbook Air 6,2

  Product ID:	0x0291
  Vendor ID:	0x05ac (Apple Inc.)
  ST Version:	1.71
  MT Version:	2.32
  Serial Number:	DQ675057JBKF94RA633
  Manufacturer:	Apple Inc.
  Location ID:	0x01000000