imbushuo/mac-precision-touchpad #40: Driver Not Showing in Macbook Air 13" Mid-2013 Model

Wonigox, 2018-01-12:
I installed the certificate to "Trusted Root Certificate Authorities" and installed the .inf file yet the driver (or Bingxing Wang) doesn't show up when updating the driver of my HID-compliant mouse (the "Show compatible hardware" is unticked).

imbushuo, 2018-01-12:
Go to device manager, select `device by connection` as the image below.

[image]

Check if your trackpad is SPI-based, like this:
[image]

Or your trackpad is USB-based:
[image]

INF file should work for USB-based trackpad (not SPI-one). If not, copy your trackpad's VID/PID (stock driver should be named `Apple Multi-touch`). Installing driver to mouse device won't work in this case as the mouse you see in device manager is a HID Top Level Collection exposed by its parent device.

Wonigox, 2018-01-12:
Okay so I got it to install (it was USB-based until Apple Multi-Touch) but now it's saying that the driver could not be started:
[image]

imbushuo, 2018-01-12:
Got it. Try removing driver manually via a backup mouse or keyboard, I will get a device later and check what happened.

Paulicat, 2018-01-29:
Just wanted to confirm the same problem as above exists on MacBook Air 2014.
Great work on this driver and looking forward to any progress you may make.

AlvaroMartinezB, 2018-02-02:
Can confirm the same issue on Macbook Pro 8.1 (Late 2011)
[image]

imbushuo, 2018-02-28:
Please check latest CI build (in release section) to see whether device can start or not. Thank you!

sirburpalot, 2018-02-28:
There is no CI build in the release section.

imbushuo, 2018-02-28:
@sirburpalot There's another issue I just found. I will tell you when it is ready.

imbushuo, 2018-02-28:
https://github.com/imbushuo/mac-precision-touchpad/releases/tag/univ-b1-i2

This release fixed the issue for some MacBook Air models. Please let me know if you still hits the issue (likely for other models). To look into the issue, I need full device ID (including `MI_` prefixed string) and device family model.

cazdli, 2018-03-28:
Bringing this back up as I own the same model.
The same error is still present on the latest release.

Apple MacBook Air laptop 6,2 13" 2013 A1466

USB\VID_05AC&PID_0291&REV_0171&MI_02
USB\VID_05AC&PID_0291&MI_02

Let me know if you need any more details.

imbushuo, 2018-03-28:
@cazdli Can you produce a trace file using the method mentioned in #62? If you are running newer CI builds, remember to replace symbol files (in `.pdb` extension) too.

cazdli, 2018-03-28:
[Bring-up.zip](https://github.com/imbushuo/mac-precision-touchpad/files/1858140/Bring-up.zip)

Here you go. Let me know if you have any issues with this.

cazdli, 2018-04-02:
[Bring-up.zip](https://github.com/imbushuo/mac-precision-touchpad/files/1866919/Bring-up.zip)

All right, done. Sorry for the delay.