imbushuo/mac-precision-touchpad #42: installation

zakhaj, 2018-01-14:
apologies if this might seem stupid.
cant install the .inf file on MBPr late 2012 10,2
repeatedly get warning of 'digital signature information'
have I missed a step?
attempted to update touchpad driver via device manager - doesn't recognise .inf file as appropriate driver...

have been eagerly awaiting a precision driver for bootcamp. long overdue!

montajd, 2018-01-14:
you need to install the certificate included in the zip. Click on it and choose manual install, and choose trusted root

zakhaj, 2018-01-14:
thanks for replying. where is it exactly? looked through everything, cant see what you mean?

zakhaj, 2018-01-14:
there's no .cer file in my zip....?

montajd, 2018-01-14:
Drivers.2014\Drivers\x64(or whatever your architecture)\ReleaseSigned\AmtPtpDevice.cer

zakhaj, 2018-01-14:
inside my zip I see:

zakhaj, 2018-01-14:
[image]

montajd, 2018-01-14:
that's the source, you need to download the actual release. 
https://github.com/imbushuo/mac-precision-touchpad/releases

imbushuo, 2018-01-16:
Directly install inf file from GitHub release should be fine (with a publisher confirmation).

closed 2018-01-16 by imbushuo