imbushuo/mac-precision-touchpad #44: paint not working !!

gastondassaro, 2018-01-17:
excellent contribution! I'm definitely thinking about making a donation. But I'm missing something to decide, I have a problem and that is using a PC that is not MAC (it's a HP desktop PC) with Windows 10 and I'm using a magic trackpad 2 and the gestures work with 3 and 4 fingers, they work click and double click ... but it does not work "paint" I can not press a finger clicking and moving another finger to select several things at the same time or to paint text ... PS: Sorry but I used a translator my language is not English. Thank you !!

gastondassaro, 2018-01-17:
PD : usé este driver "Drivers.1248.zip"

gastondassaro, 2018-01-17:
I forgot to comment that I used this driver "Drivers.2014" PD: ignore the previous comment, apart from being in Spanish I put wrong the driver that I had used, the one mentioned in that comment I could not install it in any way, the one I could install is It is called "Drivers.2014" but as I said above it does not let me select multiple elements by clicking and without dragging another finger to paint ...

imbushuo, 2018-01-17:
See #37 and #38. Improvement work is in progress. I am discuss the improvement plan with some friends, and I will post them when a final version is available. Have no ETA yet.

imbushuo, 2018-01-19:
Implemented first set of improvement in https://github.com/imbushuo/mac-precision-touchpad/commit/eeefa15b004853ed958106c5a85f3854d76dd0df. Need to further adjust additional parameters and de-fuzzing algorithm for better experience.