imbushuo/mac-precision-touchpad #45: Support for MacBook Pro Late 2013 15"

aakkam22, 2018-01-17:
Any update on when this will be supported? The 13" version is supported but not the 15" and it is not listed on the Readme.

imbushuo, 2018-01-18:
Report your trackpad's VID(vendor ID) and PID(product ID) in device manager, I will take a look.

aakkam22, 2018-01-18:
Sorry for the delay, was a little busy. Assuming that the trackpad is listed as "Apple Multi-Touch", here is the info:

[image]

imbushuo, 2018-01-18:
Will be supported after resolving one bring-up issue(being investigated now).

aakkam22, 2018-02-09:
Any update?

Let me know if you need access to any hardware.

alexmyself, 2018-02-10:
Another futur user here.. :)

imbushuo, 2018-03-01:
@aakkam22 PID 0262 should have been supported as of today. Try latest release?

closed 2018-03-01 by imbushuo

matthewdias, 2018-03-01:
works great for me 👌

aakkam22, 2018-03-01:
Just got the opportunity to test it. It's loads better than Apple's driver, but it's still not quite as good as the macOS driver. One issue with the driver is that I like to have the tip of my thumb resting on the edge of the glass while I track with my index finger and now it thinks I am pinching to zoom because of this. Didn't have this issue with macOS driver nor Apple's stock bootcamp driver. Not sure how difficult it is to address this but I'd imagine it's pretty tricky :( 

Scrolling in some applications like MS Word bring the entire window down or up when scrolling. It's hard to describe but the entire window moves while you scroll.

Despite these issues this is tremendous work you've done and I am bearing in mind that this is under development. You should be commended on your work brother. Good job.

imbushuo, 2018-03-02:
@aakkam22 Chrome will introduce Precision Touchpad support in coming releases(I've talked with Chrome folks about that), so overall experience on Chrome will be improved soon. I use a Surface Pro 4 as primary device, and these Chrome scrolling problems exist on Surface too.

I've noticed other weird behaviors as you described and working on a fix. Yes, it's tricky. I need to use some mathematical approaches to resolve this.

aakkam22, 2018-03-02:
Thanks for your reply. Yep I figured Chrome was the issue because as you've said, Chrome acts the same way on other Windows devices. 

Curious, what specifically are the 'mathematical approaches' you're working on?

imbushuo, 2018-03-03:
@aakkam22 Trackpad, like other sensor devices, do not always generate precision signals. So it is more about typical signal processing in the "mathematical approaches".