imbushuo/mac-precision-touchpad #46: Bluetooth support

dinhngtu, 2018-01-18:
I got the driver working quite well with MT2 (https://github.com/dinhngtu/mac-precision-touchpad), and I am currently looking into adding Bluetooth support. However, I don't have much background on writing Windows drivers. Could you give me a few pointers on what is needed to support Bluetooth using this driver?

imbushuo, 2018-01-18:
The Windows Bluetooth Stack is more complicated compare to USB Stack. I don't have experience with Windows Bluetooth development, but generally speaking, you need to do these things:
- Magic Trackpad 2 exposes itself as a HID over Bluetooth device in Bluetooth mode. In USB mode, you can take control of the device (bypass the native USB HID driver stack) and return required information to OS in driver. It will be much harder to do that in Bluetooth mode. If you don't want to implement a HID miniport driver, I think a HID filter driver might work(I don't know if it can intercept required HID IOCTLs or not. You need to intercept device report descriptor IOCTLs to have Precision Touchpad TLC, device configuration TLC returned to OS) 
- Understand raw input data from MT2 in bluetooth mode. Should be similar to USB one. Wellspring mode switch can be represented in USB control transfer request, and there should be a equivalent one for HID (write feature report?).
- Migrate to kernel-mode driver. I don't think implement a user-mode driver for Bluetooth mode is possible(at least now, for UMDF 2.15).

dinhngtu, 2018-01-18:
Thanks, that does seem like quite an undertaking, especially since I'm new to Windows drivers. The equivalent [Linux driver for MTP2](https://github.com/ponyfleisch/hid-magictrackpad2/blob/master/usr/src/hid-magictrackpad2-4.10%2Bhid-magictrackpad2/hid-magictrackpad2.c) uses feature reports to control the touchpad, so that might work.

> Migrate to kernel-mode driver. I don't think implement a user-mode driver for Bluetooth mode is possible(at least now, for UMDF 2.15).

What is currently blocking the use of UMDF for Bluetooth HID drivers? Microsoft does list Bluetooth as one of the available HID transports (https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/hid-transports-supported-in-windows). I'd rather the touchpad drivers stick to UMDF, since signature requirements are simpler for user-mode drivers, and does not require test-signing mode or disabling Secure Boot.

imbushuo, 2018-01-18:
@dinhngtu In this case, you probably have to bypass the Windows HID transportation stack, which is the blocking factor. Other Windows Bluetooth components will be utilized, except the native HID transportation stack since the native HID collection returned from the device is not what you want OS to know, so you have to replace it in your driver.

Filter driver probably works, depending on its location in the driver stack. I haven't tried it yet though.

dinhngtu, 2018-01-18:
I understand that the filter driver doesn't have to be a kernel driver? I might give it a try, but I'm not sure I can get it to work.

Gumija, 2019-07-26:
I have been looking into making bluetooth work for 2 weeks now, but I am still new to this and more confused than when I started. The good thing is that I think I can ask meaningful questions now.

So as I understand I can have 2 approches. 

One is that I copy the current driver which is a hid mini and switch everything that is USB to some Bluetooth code which gets extremly long and complex looking at the [bthecho sample](https://github.com/Microsoft/Windows-driver-samples/tree/master/bluetooth/bthecho).

Other is to create an Upperfilter for HIDClass and intercept the IOCTLs and I get a bit lost here. Use the hid stack to get data or again use the bluetooth stack to get the data?

On top of all this data might be different so I need to check that as well.

I already started to change the current USB implementation to Bluetooth. I didn't get very far, but I am working on it. In the meantime can you confirm that this approch is fine and I am not missing something that is way easier? Additionally could you explain what wellspring mode is? I have read that this is the new trackpad family but I don't really understand why it needs to be set/reset.