imbushuo/mac-precision-touchpad #47: Testing driver on MacBookPro11,5 (15")

shawnrc, 2018-01-23:
I've got a 15" Mid-2015 Retina MBP with the following vendor/product IDs:
```
VID_05AC
PID_0274
```

I was wondering if there was a specific branch I should check out and build if I wanted to test this driver on my machine. I'm running Windows 10 x64 1709 (OS Build 16299.192).

Also, is Visual Studio required to build drivers? I have vscode installed and the dotnet stuff running, but afaik that might not be enough.

Martin1994, 2018-01-24:
I'm trying to debug this driver on my MBA Mid14 as well. AFAIK, you need a Visual Studio (community version is fine), and the Windows Driver Kit, on another Windows machine other than your MBP. In addition, [run WDK Test Target Setup](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1) on your MMP.

However, I haven't successfully debugged it yet. My VS always disconnect from the target machine after a deployment. I wish I could have an instruction of setting up a debugging environment.

imbushuo, 2018-01-24:
Mid-2015 Retina MBP 15inch is expected to be supported. I haven't merged required changes yet.

imbushuo, 2018-03-01:
@shawnrc I am working on PID 0274. However, I need some information as I don't have such device. Could you measure your touchpad and report its width and length, in centimeters?

shawnrc, 2018-03-01:
@imbushuo sure! Grabbed a ruler and the approximate measurements are 10.5cm on the long side and 7.7cm on the short one. Let me know if you need more a more precise measure or anything else from me.

imbushuo, 2018-03-03:
@shawnrc Thanks for the info!

fenderguitar, 2018-07-20:
@imbushuo I also have a 2015 15" MacBook Pro (Macbookpro11,5)

However my Mac does not contain any SPI devices. 
[image]

I have my keyboard under the USB 3.0 Bus.

Apple Internal Keyboard / Trackpad:

  Product ID:	0x0274
  Vendor ID:	0x05ac (Apple Inc.)
  Version:	6.24
  Serial Number:	<>
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14400000 / 2
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
  Built-In:	Yes

kurtzy, 2018-09-23:
Hi @imbushuo! Just curious if you might have an ETA to have this model supported. Thank you again for this excellent work!

imbushuo, 2018-09-23:
@kurtzy ETA first week of Oct (the time when my partner can have intensive test on this model)

prashantv, 2018-12-03:
Anything we can do to help testing or to add this device?

kurtzy, 2019-01-05:
> Anything we can do to help testing or to add this device?

Happy 2019, and seconded! Happy to help with testing if needed.

borb, 2019-02-21:
Hello. I have two MacBook Pros, the first a retina 2012 (MacBookPro10,1) with a fully working PTP driver (thanks to this project).

Second is a MacBookPro11,4 with vid 0x05ac/pid 0x0274. Have tried some of the nightly builds (with mentioned IDs in the inf files) and none seem to work sadly. Code 10 when trying the driver. Would gladly offer any help/testing/information as needed.

kurtzy, 2019-03-12:
Hi @imbushuo - I hope all is well! Would love to see this model supported. I have donated to your project again. Thanks for your work on this project!

imbushuo, 2019-04-25:
See https://github.com/imbushuo/mac-precision-touchpad/pull/179

closed 2019-04-25 by imbushuo

kurtzy, 2019-04-27:
Thanks Bingxing - your driver makes such a huge positive difference and I
really appreciate your efforts. Best wishes for your health and studies!