imbushuo/mac-precision-touchpad #48: sensitivity

zakhaj, 2018-01-24:
even on the highest sensitivity under touchpad settings... on a MBPr retina late 2012....
doesn't seem to always register when touched lightly.
for example, tips of fingers isn't as responsive.  works well with the 'pads' of finger.

otherwise a fantastic piece of work and huge credit to those responsible. its great, I love it.

imbushuo, 2018-01-26:
I am thinking about exposing an interface for sensitivity control, but it is under consideration yet.

icstyle, 2018-04-12:
@imbushuo Yes, please!! Would be so great!

TGSAN, 2018-06-15:
@imbushuo Yes, using my fingertips cannot control the touchpad in MacBook Pro 2014 Mid. But BootCamp driver have no problem. ( Google Translate )

battaglia01, 2019-10-31:
Is there any way to change this?