imbushuo/mac-precision-touchpad #51: Macbook Pro Late 2011 (MacBook Pro 8.1) Device Cant Start

AlvaroMartinezB, 2018-02-02:
Tried installing it onto the Apple Multitouch, and Apple Multitouch mouse devices, neither will work. Device cannot start error 10. Mouse falls back to the windows default driver for touchpads (loosing even the basic Apple scroll)
[image]

AlvaroMartinezB, 2018-02-02:
Just realized this porblem was noticed in a previews issue thread, will close this one

closed 2018-02-02 by AlvaroMartinezB

adriankwil, 2018-11-20:
> Just realized this porblem was noticed in a previews issue thread, will close this one

could you point me to the thread which helped you solve this issue? I have the same one.
Thanks