imbushuo/mac-precision-touchpad #52: Uninstallation

halfcab1981, 2018-02-04:
Hi, I installed but having the same issues as #38 

I would like to uninstall the drivers as it is unusable at the moment. How do I do that?

Martin1994, 2018-02-04:
If you have another mouse:
Open Device Manager, find your touchpad, double click, go to Driver tab, and then click Rollback driver.

If you don't have one:
1. Literally do the same thing, but with your keyboard.
2. Windows key
3. Type Device Manager
4. Enter
5. Press tab until the device tree is selected
6. Use arrow key to select your touchpad
7. Press enter
8. Press tab until tabs are selected
9. Use arrow key to select Driver tab
10. Press Tab key until Rollback driver is selected
11. Press space

halfcab1981, 2018-02-05:
Ok cheers, that worked in that the trackpad is working again but I can no longer pair via bluetooth - it only works with the USB. Any suggestions?