imbushuo/mac-precision-touchpad #54: CI Builds

ChristopherHammond13, 2018-02-07:
Hey, 

Would it be possible to get access to the CI builds that come from the branch? I want to try some of the hid-v2 fixes but as I don't have a valid signing certificate I'd have to remove the 2014 version and install an unsigned version.

Thanks so much for everything you're doing!

imbushuo, 2018-02-08:
Approved CI builds will be published to GitHub. I will modify the badge's link later so it will point to GitHub. A task will be added so unapproved releases will be published too (but you have to take your own risk trying that, though it will not harm hardware, but a backup mouse is recommended)

ChristopherHammond13, 2018-02-08:
Amazing, thanks so much!

commit 33fad547 (2018-03-01): Point badge to public CI file release server.

This resolves #54, #57
closed 2018-03-01 by imbushuo