imbushuo/mac-precision-touchpad #58: MacBookPro8,3 (17" Late 2011) Support

kurtzy, 2018-02-28:
Greetings,

I'm following this project with great interest and will absolutely be donating to your fine work. Would love to see support for this model. There are two devices in device manager - not sure if you need both, but here is a screenshot with associated info:

[image]

Apple Multitouch:
HID\VID_05AC&PID_0252&REV_0118&MI_01&Col01

Alle Multitouch Mouse:
HID\VID_05AC&PID_0252&REV_0118&MI_02

Thanks!

imbushuo, 2018-02-28:
On-going.

khalidchawtany, 2018-02-28:
@imbushuo Is MacBookPro8,1 13" supported or considered?
Thanks

imbushuo, 2018-02-28:
@khalidchawtany Not in repository now, but MacBookPro8,x support will be added in a single batch.

montajd, 2018-02-28:
It's working on my early 2011 17" MacBook Pro. You have to update the multitouch driver, the multitouch mouse does nothing when updated

AlvaroMartinezB, 2018-02-28:
hey montajd, would you mind giving details on how you instaled the drivers? i have the same Macbook pro (13" though) and the device diddnt start

montajd, 2018-02-28:
Sure, I installed the certificate first for everyone, under trusted root. Then I updated both the "Apple Multitouch" and "Apple Multitouch Mouse" drivers under "Human Interface Devices". Reboot and you are good to go. The "HID-" drivers updated automatically after updating the multitouch drivers so do not touch those. I've attached a screenshot to show what my "Human Interface Devices" section looks like now, as well as the touchpad settings that are now enabled.
[image]

AlvaroMartinezB, 2018-02-28:
Did you install the x64 driver/certificate? 
when you updated the driver, did you browse manually for it? any specific order? 
Sorry for all the questions, its just that i tried this before and didnt work... so i want to kink our any variations on our process

montajd, 2018-02-28:
Yeah I installed the x64 driver/certificate. Install the certificate first, choose everyone, and then install it to Trusted Root, don't choose the automatic option. After you've installed the certificate, then go to device manager, right click on the drivers, choose update drivers, choose "browse my computer for driver software", browse to the Driver folder you downloaded, don't choose the inf file you first see, open the folder and choose that one. Also make sure you are using the latest release of drivers. I was facing the same issues as you with the release prior to this one.

montajd, 2018-02-28:
other than that, Maybe our hardware is different enough to where it wont work.

AlvaroMartinezB, 2018-02-28:
I guess our hardware is different enough to break it... Sad... Still getting error code 10, although i was actually able to install without disabling driver signature enforcement. 
[image]
I dont know if having trackpad ++ installed (yet, disabled) might be ruining things...

montajd, 2018-02-28:
Its only a matter of time before he gets it working for them all. Believe me, I've been waiting a while for it to work. Just make sure you are following this project so you'll know the moment there's an update. Also, The dev wasn't even aware that the latest release would work with my Macbook Pro, so now that he does, it could as simple as changing one value, one argument, whatever. And Trackpad++ sucks, its basically malware, requiring you to install another app just to use it? I don't think so.

imbushuo, 2018-02-28:
@kurtzy Try the latest driver. If it doesn't work, see #62 for collecting a trace log to me.
@AlvaroMartinezB See my reply in #62.

kurtzy, 2018-02-28:
Hi, done and received cannot start error for both devices. Followed instructions you referenced - here's the trace file: 

[Bring-up.zip](https://github.com/imbushuo/mac-precision-touchpad/files/1768564/Bring-up.zip)

Note that I first disabled/enabled USB\VID_05AC&PID_0252&REV_0118&MI_01 followed by USB\VID_05AC&PID_0252&REV_0118&MI_02. If it matters at all, I noticed the .inf file doesn't have matching strings for the hardware. Thanks again and please let me know if I can help any further!

imbushuo, 2018-02-28:
https://github.com/imbushuo/mac-precision-touchpad/releases/tag/univ-b1-i3

Please give it a try!

closed 2018-02-28 by imbushuo

kurtzy, 2018-02-28:
Genius...it works! Only had to install for MI_01 (Apple Multitouch) as well. You should write drivers for the army of awful PC touchpads out there. :)

Thank you so much!