imbushuo/mac-precision-touchpad #62: its actually working on Macbook Pro early 2011!!

montajd, 2018-02-28:
as title states, its awesome. Not perfect, but much better!!

imbushuo, 2018-02-28:
Hmmm, I didn't even know that (no device can do). May I know the device hardware ID (including `MI_` prefixed string)?

Some issue mentioned acceleration issue that I need to resolve. Some digitizers (especially Magic Trackpad 2) need additional DSP work.

montajd, 2018-02-28:
sure 
Multitouch-
USB\VID_05AC&PID_0245&REV_0108&MI_02
USB\VID_05AC&PID_0245&MI_02

Multitouch Mouse-
USB\VID_05AC&PID_0245&REV_0108&MI_01
USB\VID_05AC&PID_0245&MI_01

Honestly the sensitivity doesn't bother me at all. All gestures are currently working.
I updated both the multitouch and multitouch mouse drivers to the precision touchpad drivers. I will uninstall to find which is actually necessary to install to, but I believe it is the multitouch only.

montajd, 2018-02-28:
so I found that both the Multitouch and multitouch mouse drivers under "human interface devices" must be updated to this precision driver in order for it to work. Do not touch the existing drivers labeled "HID", they will update accordingly. The touchpad settings are showing up correctly, as well as enabling and disabling features and functions via the menu. Excellent work, seriously, this is a game-changer.

AlvaroMartinezB, 2018-02-28:
Diddnt work on my Macbook Pro late 2011 (MacBook Pro 8.1), device couldnt start code 10.
I dont really know how to find the specific device ID you asked for, but this is what i found:
Multitouch: 
USB\VID_05AC&PID_0253&MI_01\7&1a5d06a9&0&0001
Multitouch mouse:  
USB\VID_05AC&PID_0253&MI_02\7&1a5d06a9&0&0002

Tell me if you need anything else... also the way i installed the driver, i dont know if its the best. I disabled driver signature enforcement, went into the device/ update driver/ choose manually/ have disk/ then i selected the x64 folder, and installed

The driver installed but diddnt start... 
[image]

montajd, 2018-02-28:
I installed the certificate included in the zip of the latest release to "trusted root", then updated the two drivers.
[image]

montajd, 2018-02-28:
I mean how much hardware wise could have really changed between the two 2011 iterations? F-ing Apple.

imbushuo, 2018-02-28:
@AlvaroMartinezB I need your assistance to help me collect a trace file. Here's the instruction:

- Download TraceView and format bundle from [here](https://ib-deployment.s3.amazonaws.com/Staging/Trace.zip). I assume you are running the latest CI driver build.
- Decompress, you will get a directory called `Trace`. Copy it to `C:\` so you will have `C:\Trace`.
- Start `traceview.exe` as administrator, go to `Options > Configure Symbols`, directly enter path `C:\Trace\Formats` into the textbox (you don't need to select directory manually).
- Go to `File > Open Workspace`, select `Bring-up Diag`, click `OK`.
- Attempt to disable and re-enable the trackpad device. If you installed the driver on multiple devices, do it for each one and tell me the sequence (device ID).
- Right-click `Bring-up Diag` in Group ID list, click `Stop`.
- You will see a big file ends with extension `.etl` in `C:\Trace`. Compress and send it to me for further diagnostics.

AlvaroMartinezB, 2018-02-28:
Okay just, What is the CI driver? i have no clue if i have the latest version... i dont know what it is
Do you want me to do this with you driver installed? or it doesent matter? (because im  unable to use the trackpad with it installed, would have to fetch a mouse)

AlvaroMartinezB, 2018-02-28:
[Bring-up.zip](https://github.com/imbushuo/mac-precision-touchpad/files/1768253/Bring-up.zip)
Here you go, hope thats what you needed. Tell me if you need anything else

closed 2018-02-28 by imbushuo

imbushuo, 2018-02-28:
@AlvaroMartinezB https://github.com/imbushuo/mac-precision-touchpad/releases/tag/univ-b1-i3

IronVanadium, 2018-03-01:
Works for me too now with Mid-2012 13-inch MacBook Pro. I now have the precision touchpad settings, yay! Precision scrolling works great, but it's still quite slow (especially in browsers).  I have to scroll down like 10 times just to get to the bottom of a page. Any way to increase the acceleration? Bumping up the number of lines scrolled at a time in More Touchpad Settings makes scrolling faster, but then I seem to lose a lot of the precision.

imbushuo, 2018-03-01:
@IronVanadium I would like to evaluate how slow it is to determine the situation. Can you upload a video demonstrate the scrolling?

AlvaroMartinezB, 2018-03-02:
Hey! I can confirm it working now, but I agree that scrolling is a bit slow... Its sometimes confused with a pinch, and sometimes not detected. If you swipe quickly and let go, it will continue moving quickly as i showed, but if you try to scroll more precisely, its slow.

[its actually working on Macbook Pro early 2011!! · Issue #62 · imbushuo_mac-precision-touchpad - Google Chrome 3_1_2018 7_34_10 PM.zip](https://github.com/imbushuo/mac-precision-touchpad/files/1773473/its.actually.working.on.Macbook.Pro.early.2011.Issue.62.imbushuo_mac-precision-touchpad.-.Google.Chrome.3_1_2018.7_34_10.PM.zip)

IronVanadium, 2018-03-03:
@imbushuo The video AlvaroMartinezB posted demonstrates it perfectly. Just seems to take a lot of scrolling to move through a long page. It's a bit better in Edge, but in Chrome and especially FireFox it's quite slow.

commit e4bc6520 (2018-02-28): Add Wellspring 5A support.

Wellspring 5A is identical in controller perspective. Full support has been implemented to resolve #62, #58, #50.

Need to revisit #41 again.