imbushuo/mac-precision-touchpad #65: Trackpad dead after resume from hibernation

aakkam22, 2018-03-03:
The trackpad is dead after resume from hibernation. Not working at all, even after several minutes. Trackpad is ON in Settings. 

Macbook Pro 11,2 

[image]

imbushuo, 2018-03-03:
Please follow #62 to generate a trace log. (Remember to replace the pdb symbol file if you are running the latest build). Thank you!

Pajn, 2018-10-05:
I have the same problem, device manager says "Windows has stopped this device because it has reported problems. (Code 43)"

Here is a trace, though after the problem. However when disabling/enabling the device for the trace I got a power failure (Code 10) instead.

[Bring-up.zip](https://github.com/imbushuo/mac-precision-touchpad/files/2449058/Bring-up.zip)

imbushuo, 2019-10-21:
For SPI trackpad, please go to #203. The trace doesn't actually provide useful information