imbushuo/mac-precision-touchpad #66: [Note] Next stage driver plan

imbushuo, 2018-03-05:
Here's a list of things to do for the next stage. This issue will be split into several projects later.

# SPI trackpads
Apple SPI device drivers use Windows Simple Peripheral Bus Extension Driver model, so writing a driver for SPI-based devices should not be too difficult (WDF provides everything). And a good news is we don't need a KMDF driver for SpbCx peripherals. Only SPB controller needs that and Intel has done everything for us.

# S3/S4 wake
This issue has workaround implemented during first MacBook Pro port. I have no idea why it doesn't work on certain models. If you hit this issue, capture a trace log file and send it to me for further diagnostics.

# HID descriptor library
As more models are going to be supported, hard-coding HID descriptors is not a sustainable way from maintenance's perspective. All HID descriptors that report to OS should be generated from a library. This library will accept a list of HID attributes and builds an HID descriptor automatically.

# Shared library and difference drivers
SPI-based models will not share the driver with USB-based models, as they have different IO targets and Windows Driver Model has no support for such hybrid scenario AFAIK. But they will share most parts of Wellspring logic control and upper HID services. Shared components should be moved into a separate static library.

# Signal processing
Mathematical approaches will significantly improve the experience.

montajd, 2018-03-06:
This driver has completely changed the Bootcamp experience, keep up the good work!

kurtzy, 2018-03-09:
100% agree, also love the updated background story...great motivation!

MilanBaca, 2018-03-12:
I am agree - in bootcamp is greate :-)
but i have to ask, is there a priority to make MT2 more stable? I have replaced mouse with MT2 on my other computer with Windows10 but working with it is not so good as with touchpad on my macbook.

myonlylonely, 2018-03-16:
May I ask when will the driver support MacBookPro11,3 (Mid 2014)? Is it in the plan?

imbushuo, 2018-03-24:
@myonlylonely Yes

imbushuo, 2018-07-07:
See #88.

closed 2018-07-07 by imbushuo