imbushuo/mac-precision-touchpad #70: Regular PC with MT2 and BT

carbonatd, 2018-03-30:
Could not install these drivers after pairing the device through bluetooth.  I am using a non-Apple PC desktop with a BT USB adapter.  I am using other drivers from MagicUtilities, which seem to work, but they are not very smooth..  After manually loading the driver for the HID Apple Multi-Touch Pro in the device manager and restarting, I lose all connectivity.  I read in another post the bluetooth might not work?  Is this correct?

[image]

shanemadden, 2018-04-24:
I'm seeing the same - Windows seems to think the driver isn't valid for the device when it's paired over bluetooth (mine is a motherboard-integrated bluetooth radio), while it works great when the MT2 is connected directly to USB.

imbushuo, 2018-07-19:
Research is started (seems similar to the SPI driver). I will post the result next week.

closed 2018-07-19 by imbushuo