imbushuo/mac-precision-touchpad #79: macbook pro 2017

jx3fans, 2018-05-13:
您好，macbook pro 2017 无bar款安装驱动后似乎没有作用，翻看inf后推测可能是硬件id不对。
我不懂win下的驱动开发，不知道下图内容对您是否有帮助
[image]
[image]

英语不好，只好用中文，见谅。

jx3fans, 2018-05-18:
囧，因为不支持spi呀，作者你按你自己想法慢慢弄，等啥时候有spi模式的驱动了，我可以帮你测试测试😄

imbushuo, 2018-07-14:
SPI for recent MacBook Pro will be implemented very soon.

closed 2018-07-14 by imbushuo

imbushuo, 2018-07-15:
As of today, support for MacBookPro14,x has been merged into master branch.

alanturing7, 2018-07-23:
@jx3fans 您好，请问你问题解决了吗？我也是 mbp 17，希望得到你的分享 :)

jx3fans, 2018-07-24:
@alanturing7 目前还没，大神还在为新spi驱动申请whql的认证，还没放出编译之后签名好的驱动，目前申请到那个步骤了还无从知晓。有些人是自己下载源码后自己编译的，这个需要在win下开启驱动测试模式的。我这边没下载这套编译环境，于是就没测试。

alanturing7, 2018-07-24:
@jx3fans 好的，谢谢！我会持续关注，如果有更新也望能够告知！