imbushuo/mac-precision-touchpad #82: Uninstall back to default driver

seanho12345, 2018-05-27:
I install your driver and tried a little bit. But I still prefer the original one.
How can I get back to the original driver that apple gave.
p.s: rollback driver is greyed out. 
      already tried to install driver again from the bootcamp installation usb.
     (WINDOWS_SUPPORT folder)

jamesh53, 2018-06-07:
Same question. I would like to uninstall it. Thanks!

TMAoyagiKosei, 2018-06-12:
I have a same question with them. I'm in troubled because of a bug in the Drag&Drop.

YeshYyyK, 2018-06-17:
Go to device manager and disable the Precision Touchpad from there, it should revert to the Bootcamp one. Restart later on. 

@seanho12345 
@jamesh53 
@TMAoyagiKosei

jamesh53, 2018-06-19:
Has anyone tried this? I’m a bit nervous to try it without confirmation. Thanks!

YeshYyyK, 2018-06-20:
The settings won't exactly be the same, but it does seem it work

seanho12345, 2018-06-20:
It just become a normal track pad.
There is no gesture. 
Does anyone know to get back the default bootcamp gesture?

jamesh53, 2018-06-22:
Maybe it would work if you just reinstalled Bootcamp onto windows? I still haven't tried uninstalling yet. You could also try installing something else like trackpad++? Just some ideas. Let me know how it goes I am curious.

seanho12345, 2018-06-22:
I successfully get it back to bootcamp driver but you'll need mouse for it.
1.go to device manager
2.look for Apple precision track pad
3.update the driver to "USB input device"
 (Your trackpad will stop working)
4.get your bootcamp disk or USB which you create for the windows installation.
5.go to WINDOWSSUPPORT folder
6.run the bootcamp program and repair
Then it should work!
Good Luck :)

closed 2018-07-17 by imbushuo