imbushuo/mac-precision-touchpad #83: Cool project, any to help with Magic Trackpad2 (working on a Surfacebook 2)

Din0Git, 2018-05-30:
Hi, 

I recently found this project and i tried it on a magic trackpad 2. It is a really cool project (i know it is unstable now but it's still cool). Is there any way I can contribute. I have a magic trackpad 2 and a surface book 2 with its build-in trackpad from MS.

Let me know
Dino

AkashGutha, 2018-08-02:
@imbushuo I would like to contribute too. Can you provide links to apple drivers or the procedure to get them so that anyone can read and learn from them?
Also, is there any slack or gitter channel where we can discuss?

imbushuo, 2018-09-03:
@ynjiun Thanks. The issue is:

- Magic Trackpad (as well as other Apple trackpads) report touch contact point size and probably pressure information
- Reading on MT2 is not really reliable
- Windows just wants two bits: one contact status (finger on trackpad or not), the other is confidence (whether this is a finger or not)

What we have to address is, properly convert Apple reading into Microsoft data.

m4gr4th34, 2019-06-15:
@ynjiun , hope you can help out.  My poor MT2 lies sadly unused on my desktop.