imbushuo/mac-precision-touchpad #89: How to set up testing environment

pieterdonkers, 2018-07-09:
I would like to start helping with this project.
How would I install the drivers without permanently overriding my normal drivers?
I know that you can reinstall the old drivers but this process seems annoying every time I want to test.

I own a 2014 macbook pro retina 13 inch and have some experience within software engineering and electrical engineering. So thing like SPI are natural to me.

Martin1994, 2018-07-09:
I wonder if I can debug a user-mode driver on a single computer instead of two.

imbushuo, 2018-07-09:
SPI needs KM mode, and for some reason, tracing is not working yet (I am looking into it), you have to use DebugView or attaching a kernel debugger.

A way to test km driver safely requires two computer, one as the debugger and another as the debuggee. In this way, test-signing is not required (Windows permits unsigned driver load when KD is attached).

pieterdonkers, 2018-07-09:
Ok, thnx for the quick response.

closed 2018-07-14 by imbushuo