imbushuo/mac-precision-touchpad #90: Magic Trackpad 1

JannesMeyer, 2018-07-09:
Is there a technical reason why the first Magic Trackpad can't be supported? I don't see any mention of it anywhere in this project.

Or is it just because you don't have the hardware and time available?

imbushuo, 2018-07-09:
Since it's a Bluetooth HID device, it might be tricky to intercept HID IOCTLs and make them PTP. I do have a device, but I have no idea as well as time available.

closed 2018-07-09 by imbushuo

FernanDK, 2019-05-07:
has this issue been addressed or is it still incompatible? 

I have a magic trackpad 1 and am wondering if I should buy the 2nd gen or wait for the 1st to be supported?

imbushuo, 2019-05-08:
MT1 is currently not on the plan.

Thanks,
Bingxing Wang

JannesMeyer, 2019-10-06:
Do you have any suggestions how someone would get started implementing this?

I have C and C# knowledge but have never worked with anything related to drivers or Bluetooth.