imbushuo/mac-precision-touchpad #96: MacBook Pro Touch Bar 2016 Support

Kent-Narun, 2018-07-17:
Can make the driver for MacBook Pro Touch Bar 2016; The devices ID is attached as follow:
[image]
[image]

imbushuo, 2018-07-17:
`0x0277` is included, which means the SPI driver already implemented support.

https://github.com/imbushuo/mac-precision-touchpad/blob/702e8c518ab052272c2f5f62c33d12a76eb160e2/src/AmtPtpDeviceSpiKm/AmtPtpDeviceSpiKm.inf#L41

The current SPI driver is test-signing (and no binary provided yet), which means you have to enable test signing to install the driver.

Kent-Narun, 2018-07-17:
Ok so I will wait for the binary then. here is my the tracpad id in windows 10 "SPI\VID_05ac&PID_0277&MI_02"

imbushuo, 2018-07-17:
@AndrewOCC 

https://github.com/imbushuo/mac-precision-touchpad/blob/702e8c518ab052272c2f5f62c33d12a76eb160e2/src/AmtPtpDeviceSpiKm/AmtPtpDeviceSpiKm.inf#L43

This is included in current implementation too. Currently we are waiting for #95, should be cleared in 2 weeks. (The blocking factor are Dun & Bradstreet and DigiCert)

AndrewOCC, 2018-07-17:
Great news - congrats on getting to the home stretch

monoKeith, 2018-07-19:
MacBook Pro late 2016 no-touchbar

Model Identifier:	MacBookPro13,1

Product ID:	0x0276
  Vendor ID:	0x05ac (Apple Inc.)
  ST Version:	8.96
  MT Version:	4.69
  Manufacturer:	Apple Inc.
  Location ID:	0x01000000

imbushuo, 2018-07-19:
@monoKeith 

https://github.com/imbushuo/mac-precision-touchpad/blob/702e8c518ab052272c2f5f62c33d12a76eb160e2/src/AmtPtpDeviceSpiKm/AmtPtpDeviceSpiKm.inf#L40

Supported too.

Rastafabisch, 2018-07-20:
Here you got all the relevant information from my model, too. I hope it helps!
Thanks for your efforts! Cheers

[image]
[image]

imbushuo, 2019-10-21:
0278 should be supported per https://github.com/imbushuo/mac-precision-touchpad/blob/702e8c518ab052272c2f5f62c33d12a76eb160e2/src/AmtPtpDeviceSpiKm/AmtPtpDeviceSpiKm.inf.

closed 2019-10-21 by imbushuo