imbushuo/mac-precision-touchpad #100: Force Touch

imbushuo, 2018-07-19:
Force touch exposes additional HID collections to the host OS (accelerometer and actuator). On USB-based trackpad, they are recognized by default, but no OS support is implemented. On SPI-based trackpad, another miniport driver is required to make it available to host OS.

See if we have applicable scenarios for Force Touch on Windows.

imbushuo, 2019-08-28:
Will be tracked in [AB#635](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/635).

imbushuo, 2019-10-09:
Force touch structure:
14-bytes packed structure
```
typedef struct _HAPTIC_FEEDBACK {
    USHORT Magic;
    UINT Strength;
    int X;
    int Y;
} HAPTIC_FEEDBACK, *PHAPTIC_FEEDBACK;
```

It exists as a dedicated HID device on MI_03 (Magic Trackpad 2) or MI_02 (others). Tap test works fine but I don't have idea about usable scenarios on Windows.

DrJedikiah, 2020-01-09:
It could be used as right click, folder&file rename, middle mouse button for opening in new tab etc.

chanibal, 2020-01-20:
Middle click +1 this would solve a lot of my issues. Middle-button drag would work (used in my uml diagraming software), I could also use the same functionality as in macos with middle clicking links to open in new tabs.

Is more than one button force touch recognizable? Ex. can you distinguish between one button force touch and two or three button force touch?

zzFluke, 2020-11-01:
Hey @imbushuo, looked at your feedback test program but couldn't understand how to get accelerator reports from trackpad with applied force values. Is it possible? Could you please point me to the place where I can read about that?

imbushuo, 2020-11-02:
@zzFluke Force value retrieved from touchpad HID collection, the accelerator seems unknown to me. Actuator is used for haptic.

not-holar, 2021-12-15:
Would it be possible to add a setting for how hard you need to press for a click to register? It feels a little too stiff on my MT2, fine for clicks, but a little difficult to drag while holding