imbushuo/mac-precision-touchpad #99: Sleep wakeup issues (SPI + USB)

imbushuo, 2018-07-19:
# SPI
- Probably Bugcheck `UNEXPECTED_KERNEL_MODE_TRAP`.
- Cursor lost (observed in S3, not S0ix).

# USB
- Randomly disabled.

See #71, #65.

Kent-Narun, 2018-08-09:
My Macbook Pro 2016 always have unexpected_kernel_mode_trap after wake up from sleep.

imbushuo, 2018-08-11:
Aware of the issue. This can be reproduced on New MacBook (2016).

cubesu9ar, 2018-09-27:
reproduced on my new macbook 2015, bluescreen while resuming from hibernate, stopcode is UNEXPETED_KERNEL_MODE_TRAP

imbushuo, 2018-09-27:
@cubesu9ar A recent build should fix this (but introducing another non-critical issue). What's your build version?

cubesu9ar, 2018-09-27:
Can you share the recent build's URL please? I have this issue with SPI driver version 14.56.52.89

imbushuo, 2018-09-27:
@cubesu9ar You have to build this as binary is not published yet. The change is located at https://github.com/imbushuo/mac-precision-touchpad/commit/5917e2fbfc4674d5fe5b9c1130130d144094b2df.

cubesu9ar, 2018-09-27:
I made a build on my PC, with the patch you mentioned. And installed it to my NMB, will feedback if any issues found.

If anyone want the build, can download here
[AmtPtpDeviceSpiKm.zip](https://github.com/imbushuo/mac-precision-touchpad/files/2424151/AmtPtpDeviceSpiKm.zip)
It signed with my signature, install w/o import signature also works on my side, only cause some warning

imbushuo, 2018-09-29:
@cubesu9ar Any issues (including the cursor lost, at least I heard of)?

kazenoai, 2018-10-01:
The cursor will appear to screen bottom when I touch the top track pad. I look seem cursor lost. I don't know is this issue?  I also meet unexpected UNEXPETED_KERNEL_MODE_TRAP issue when the computer resume from hibernation.
Computer: Macbook 2015 13.3
OS: Windows 10 1803
SPI driver.

imbushuo, 2018-10-01:
@kazenoai Try recent builds (there’s one below)

kazenoai, 2018-10-02:
I have builded the recent software. And try it in my computer. The unexpected kernel mode trap has been resolved. Thanks!

Pajn, 2018-10-02:
When I try to install the linked driver I get "a problem was encountered while attempting to add the driver to the store", does anyone know how to fix that?

cubesu9ar, 2018-10-06:
@Pajn Have you tried to set "testsigning on", it solve my "Code 52" problem

`bcdedit /set {current} testsigning on`

cubesu9ar, 2018-10-06:
Driver startup failure occurs , with "Code 43"
`由于该设备有问题，Windows 已将其停止。 (代码 43)`

The device can be recovered by deleting driver and rescan hardware
New Macbook 2015, SPI device

tonygrue, 2019-02-01:
@imbushuo With the SPI driver (build it from source and installed it with test signing on) I'm hitting issues where the cursor is dead sometimes when waking from sleep or powering on; not always but some of the time. In device manager it says the device stopped working due to an error and disabling and re-enabling the device, fixes the issue.  I turned on tracing and see this:

"00000001	AmtPtpDeviceSpiKm	4	5812	3	1	01\31\2019-18:59:25:404	AmtPtpSpiSetState WdfIoTargetSendIoctlSynchronously failed with 0xc0000001(STATUS_UNSUCCESSFUL)"

I'm familiar with Windows / Win32 as a developer, but have no real experience making a driver, so while I'm going to poke around a bit, it's very unlikely i'll figure it out myself, so I'm posting here as an fyi.

I'm on a mid 2017 macbook pro, 13", no touchbar.

tonygrue, 2019-02-01:
Another wakeup scenario that failed for me just now, but with different log output, is my computer ran out of battery and so windows turn off the laptop.  When it started up, the trackpad didn't work.  TraceView started up, and didn't capture all the events, but it did get this event 70K plus times:
"00171519	AmtPtpDeviceSpiKm	4	6332	3	171519	02\01\2019-01:27:43:172	AmtPtpRequestCompletionRoutine No pending PTP request. Routine exit"

eventually it ended in "00176118	AmtPtpDeviceSpiKm	4	6332	3	176118	02\01\2019-01:27:43:610	AmtPtpDeviceSpiKmEvtDriverContextCleanup Entry"

I'm not sure if that happened in response to be disabling the device in Device Manager, or if it happened on it's own right before I disabled it. 

Once I restarted the device, everything is up and running.

Anyway, just an FYI so you have this additional situation.

imbushuo, 2019-02-13:
@tonygrue The issue I am currently aware is that it might not follow a proper D3->D0 sequence all the time in S4 situation (likely a power request failed?). I am looking into possible workarounds.

jvandenbroeck, 2019-02-14:
> My Macbook Pro 2016 always have unexpected_kernel_mode_trap after wake up from sleep. The driver version is attached below...
> MacbookPro-Touchpad-PrecisionDriver-2016-2018.zip

I believe I have the same issue, I'll try the most recent driver one of the next days to see if it's resolved;  fyi bugcheck https://pastebin.com/gG8zm2nf

LukasMendez, 2019-02-19:
> > My Macbook Pro 2016 always have unexpected_kernel_mode_trap after wake up from sleep. The driver version is attached below...
> > MacbookPro-Touchpad-PrecisionDriver-2016-2018.zip
> 
> I believe I have the same issue, I'll try the most recent driver one of the next days to see if it's resolved; fyi bugcheck https://pastebin.com/gG8zm2nf
> 
> Edit: after building the latest release from source everything seems to be working perfectly! Haven't experienced any issues:)

What did you actually do? I've installed the latest release (12 Jul 2018) by following these instructions:

https://www.youtube.com/watch?v=C9x9zwMalXU&feature=youtu.be

Mine is also a Macbook Pro 2016. Everything is working fine, its just whenever I wake it up after sleepmode, it crashes and restarts the computer.

jvandenbroeck, 2019-02-19:
I build the latest release from source, (visual studio+ WDK) the latest release is from november 2018. If you search in the issues there are more comments listing how to install the SPI driver & a very brief comment on what's required to build.

LukasMendez, 2019-02-19:
But did you manage to fix the sleep/wake issues?

jvandenbroeck, 2019-02-19:
yes, haven't had the issue since I updated the driver (for 2-3 days now)

LukasMendez, 2019-02-19:
Damnit.. I don't have either Visual Studio or WDK. There's no easier way to install it from the source code?

AuroraWright, 2019-02-26:
Can confirm the sleep/wake random malfunction, compiled from source and using on October update

ravenxiao454, 2019-06-09:
Macbook pro 13寸2015 也有这个问题
用键盘（电源键以外）或触控板唤醒睡眠状态会触发驱动失效
外接鼠标或者电源键唤醒就不会触发驱动失效
希望这能帮到你