imbushuo/mac-precision-touchpad #105: Cant Install Driver

PwnedBOT, 2018-07-24:
Hey,

I cant install the driver. I followed the instructions, but it does not show up as a precision trackpad...

johnothwolo, 2018-08-10:
Where is the problem exactly? Does it prompt you to install the driver? Does it install? Which MacBook model is it?

PwnedBOT, 2018-08-11:
I can install the Driver. But it doesn‘t seem to be active because it does not show up in settings. (the touchpad-menu).
I have a MacBook Pro 11,3. (Bootcamp-Software is installed.)

Kent-Narun, 2018-08-21:
1. Unpack the zip.
2. Install "AmtPtpDevice.cer" certificate file to your machine's root authority store by right click on the file and select install.
3. Go to "AmtPtpDevice" folder.
4. Right click AmtPtpDevice.inf and select install.
5. Go to Device Manager>Human Interface Devices
6. Right click Apple SPI Trackpad and select "Update Driver">"Browse my computer for driver software">"let me pick of a list of available driver on my computer"
7. Select Apple SPI Precision Touchpad Device.
[Macbook Pro 2016 WinPrecisionTouchpad.zip](https://github.com/imbushuo/mac-precision-touchpad/files/2304825/Macbook.Pro.2016.WinPrecisionTouchpad.zip)

PwnedBOT, 2018-08-25:
Yeah but the Problem is, that when I wanna update my driver, windows says that currently, there is a better driver installed, so windows can't update the driver...

PwnedBOT, 2018-08-25:
I have Windows 10 Version 1803

PwnedBOT, 2018-08-25:
Edit: My Trackpad does not show up as a SPI-Trackpad. Instead it is listed under USB Devices in the System Profiler in macOS.

sergiojavdiaz, 2018-09-16:
Hey bro, follow the instructions of the author but install the Drivers.2358.zip. Tell me if it worked for you, cheers

PwnedBOT, 2018-09-16:
It worked! Thank you very much.

One last Question: does the Driver also work on a Magic Trackpad 1?
cheers

sergiojavdiaz, 2018-09-16:
You're welcome.
As far as I know it works with Magic Trackpad 2 only (I'm not the developer).

duggie1031, 2020-01-23:
Does this work on a Windows 10 machine (not apple)? 
I have the Apple Magic Trackpad 2 and trying to follow install instructions but not working

nwittwer, 2020-01-24:
1. Download latest release here: https://github.com/imbushuo/mac-precision-touchpad/releases 
(in my case it was "Drivers-amd64-ReleaseSigned.zip" for Windows 10)
2. Right click the downloaded .ZIP and select "Extract all..." (don't just double-click!)
3. Open the directory has been extracted
4. Right click `AmtPtpDevice.inf` > Select "Install"
5. Accept the security pop-ups
6. Open Device Manager (Windows > Device Manager)
7. Navigate to "Human Interface Devices"
8. You should find an item like "Apple USB Precision Touchpad Device (User-mode)". 
_NOTE: It may be named slightly differently if you're using it with Bluetooth_

duggie1031, 2020-01-25:
Thanks very much for this, works perfect, got scrolling on my W10 machine now. Only when plugged in and not on Bluetooth but thats not a problem

Thanks again nwittwer

pgumiela, 2021-02-11:
@imbushuo, I think we can close this one :)

closed 2021-02-11 by imbushuo