imbushuo/mac-precision-touchpad #107: MacBookPro13,2

jourdant, 2018-07-29:
MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports)

**Apple Internal Keyboard / Trackpad:**
Product ID:	0x0277
Vendor ID:	0x05ac (Apple Inc.)
ST Version:	8.96
MT Version:	4.69
Serial Number:	FM764720255GJCQxxxxxx
Manufacturer:	Apple Inc.
Location ID:	0x01000000

BrianValente, 2018-07-29:
The driver is already working on the MacBook Pro 2016.

jourdant, 2018-07-29:
Awesome - thank you. To install, do I just replace the default HID driver by manually selecting the .inf file?

BrianValente, 2018-07-29:
I did this: Update driver > Browse my computer for driver software > Let me pick > Have disk... > Select inf > Install.
Then the device will show as "Apple SPI Precision Touchpad Device".
Remember to first enable test mode on Windows. Anyway, I recommend you to wait for a signed build, enabling test mode can be dangerous and some multiplayer games won't let you play.

closed 2018-07-29 by jourdant