imbushuo/mac-precision-touchpad #109: Signed SPI driver

enricmarques, 2018-07-31:
Hi, 

I've a MacBook 12" 2017 (Id: MacBook10,1, Model A1534).

I've successfully compiled the SPI driver, but I cannot install it without enabling Test-Signing in my computer. I've also tried the SPI driver from the CI builds, with same result. It seems it is signed with the WDK test signature.

I use the laptop inside our organization, and I'm not allowed to enable test-signing.

I've tried to enable test-signing to try the SPI driver, and it works perfectly, but I only used it a couple of minutes. Are you going to release a signed SPI driver?

Thanks!

1132719438, 2018-08-01:
Can you share the compiled SPI driver?

enricmarques, 2018-08-01:
Hi @1132719438 ,

You can get it from the continuous integration builds:
https://ptp-release.imbushuo.net/

[image]

Enric

ww123td, 2018-08-01:
Hi @enricmarques The CI builds seem to have not been updated to date and made my trackpad recognized as a touchscreen (which is an already resolved issue). I don't have access to my workstation so I cannot compile it myself. It would be great if you can share yours so I can make mine working again.

Thanks!

enricmarques, 2018-08-01:
Last CI build is from July 14th, which is when the last commit was done on the repository. My build will be identical.

ww123td, 2018-08-01:
Hi @enricmarques the last CI build is 3ba51c8, which was the second commit of the day. The author then followed up with a series of fixes including the bug that identified the touchpad as a touchscreen. Again, if you are willing to share your build that would be super helpful, thanks!

enricmarques, 2018-08-01:
Hi @ww123td,

Here is the driver, built on my computer yesterdar, that worked fine on my MacBook 12" 2017:

[AmtPtpDeviceSpiKm.zip](https://github.com/imbushuo/mac-precision-touchpad/files/2250257/AmtPtpDeviceSpiKm.zip)

Enric

1132719438, 2018-08-02:
This build is from the latest commit, and it works on MacBook 15'' 2017

https://ptp-release.imbushuo.net/Apple%20PTP%20Trackpad%20Build/702e8c518ab052272c2f5f62c33d12a76eb160e2/

1132719438, 2018-08-02:
It seems Swipe left or right with two fingers to show the previous or next page not works.

vdroid711, 2018-08-02:
Thanks guys! 
All windows gestures are working perfectly!

Kent-Narun, 2018-08-08:
Thanks guys. Now the driver show up in device manager for MacBook Pro 2016 🙏🏼🙏🏼🙏🏼...

Kent-Narun, 2018-08-09:
@enricmarques I use your driver but my computer have blue screen after wake up from sleep (unexpected_kernel_mode_trap after wake up from sleep). Do you have any fix for this? I'm using Macbook Pro 2016 touch bar (13,2).

enricmarques, 2018-08-09:
Hi @Kent-Narun,

I’m not the developer of this driver, just tried it. You should create a new issue so the project developer can take a look.

Enric

imbushuo, 2018-08-13:
@Kent-Narun This is a known issue (see the wakeup thread).

imbushuo, 2018-08-13:
The EV-signed driver will be available about a month later (well, the procedure is far more complicated than a normal signed driver).

phu54321, 2018-10-08:
Is there any ETA for signed driver?

imbushuo, 2018-11-29:
@xeijin We have some issue on entity presence validation and is currently working on that.

steven-ray, 2019-03-01:
Hmm, can's seem to get the SPI driver loaded for the device.  Have MacBookPro14,1 and downloaded a few versions above.  Verified the entry in the .inf file.  Installed certs and .inf.  Enabled testsigning on.  Rebooted the system and then try to update the driver by browsing to the driver location.  Windows just says I have the best driver installed and will not update.  Any thoughts?

xeijin, 2019-03-01:
@imbushuo any updates? Still uncomfortable with leaving my computer in test mode given security risks...

imbushuo, 2019-07-30:
SPI drivers are signed as of today.

closed 2019-07-30 by imbushuo