imbushuo/mac-precision-touchpad #110: MacBook Pro 9,2 installation

yihao96, 2018-08-03:
So I've installed the certification to Trusted Root Certification Authorities, installed the AmtPtpDevice.inf in the AmtPtpDevice folder. Both installation success but Apple Precision is not showing in device manager?

Kent-Narun, 2018-08-21:
1. Unpack the zip.
2. Install "AmtPtpDevice.cer" certificate file to your machine's root authority store by right click on the file and select install.
3. Go to "AmtPtpDevice" folder.
4. Right click AmtPtpDevice.inf and select install.
5. Go to Device Manager>Human Interface Devices
6. Right click Apple SPI Trackpad and select "Update Driver">"Browse my computer for driver software">"let me pick of a list of available driver on my computer"
7. Select Apple SPI Precision Touchpad Device.
[Macbook Pro 2016 WinPrecisionTouchpad.zip](https://github.com/imbushuo/mac-precision-touchpad/files/2304824/Macbook.Pro.2016.WinPrecisionTouchpad.zip)

yihao96, 2018-08-21:
Mine is USB, not SPI

yihao96, 2018-08-21:
issue solved by using this release [https://github.com/imbushuo/mac-precision-touchpad/releases/tag/univ-b1-i3](url)

closed 2018-08-21 by yihao96