imbushuo/mac-precision-touchpad #119: new MacBook Early 2015

cubesu9ar, 2018-09-03:
MacBook8,1
Touchpad is a SPI bus device

VID:05AC
PID:0272

current driver seems not work on my new macbook, force install makes  touchpad stop working

imbushuo, 2018-09-08:
Any chance to log kernel messages using DbgView?

jonalbo, 2018-09-08:
Gonna be honest, no clue what that means hahaha. If you tell me what to do I can do it. I was able to run it and everything on windows but once it installed it wouldn’t use the driver or give me the new mouse options. I can send you whatever info from either side, just let me know!

imbushuo, 2018-09-10:
1. Get this tool: https://docs.microsoft.com/en-us/sysinternals/downloads/debugview
2. Run as administrator
3. Enable kernel logging (in menu)
4. Load the driver

Save the kernel log as a file, and send it to me.

cubesu9ar, 2018-09-16:
[GRANT-MACBOOK.LOG](https://github.com/imbushuo/mac-precision-touchpad/files/2386543/GRANT-MACBOOK.LOG)

Attachment is the debugView logs from my nmb.
What I did:
1. start up debugView with administrator
2. start device manager
3. Find "Apple SPI touchpad" , right click, select "upgrade dirver" "Browse my computer for driver software">"let me pick of a list of available driver on my computer"
4. Select "Bingxing Wang" and "Apple Precision Touchpad Device"

imbushuo, 2018-09-16:
I don't see anything useful there, would you like to set up a remote debug session with me later this week?

cubesu9ar, 2018-09-20:
I tried again with SPI signed driver from [Here](https://github.com/imbushuo/mac-precision-touchpad/issues/109#issuecomment-409779286)
But still not work, seems I installed without correct certification. 
Sorry for Chinese  UI, I believe you can read Chinese, so not translate them :-)
[image]

But I can found the certification as below
[image]

imbushuo, 2018-09-20:
@cubesu9ar So I am going to reply in Chinese too :)

Kernel-mode 驱动需要 EV 签名，User-mode 驱动不用，所以 USB 版和 SPI 版的签名情况不同（新机器带 T2 的有 Secure Boot 了，还需要 WHQL/Attestation 跨签，但是我们先不关心这个），但是我还在和我朋友走 EV 的流程，还需要一段时间。目前你需要打开 Test Signing 来安装：

```
bcdedit /set {current} testsigning on
```

然后你再试试。

cubesu9ar, 2018-09-20:
Thanks for quick response

It works on my macbook with your guide(to enable test signing)

closed 2018-09-20 by cubesu9ar

hoangvuit, 2019-05-04:
I have a Macbook Pro 15'' Mid 2015 and it does has an USB one. Tried with USB version of this driver but still no luck (also turn on testing mode). Can you guys help me?