imbushuo/mac-precision-touchpad #123: Confusing installation process (mid 2012 macbook pro)

Adamsapplespie, 2018-09-12:
So I believe I got the driver installed following Kent-Narun's instructions;

1. Unpack the zip.
2. Install "AmtPtpDevice.cer" certificate file to your machine's root authority store by right click on the file and select install.
3. Go to "AmtPtpDevice" folder.
4. Right click AmtPtpDevice.inf and select install.
5. Go to Device Manager>Human Interface Devices
6. Right click Apple SPI Trackpad and select "Update Driver">"Browse my computer for driver software">"let me pick of a list of available driver on my computer"
7. Select Apple SPI Precision Touchpad Device.

However I cannot find the driver itself and honestly have no clue what I'm doing. If there is some kind of guide or video tutorial regarding how to install this driver that would be amazing, because it seems there is very limited information atm. It's apparent too judging by how many people's comments on the demo video are simply them not understanding how to install the damn thing.

Sorry to sound critical, but this should be essential for people to install the driver correctly and with ease.

NotStewpidAidan, 2018-09-13:
You shouldn't install the driver by right-clicking the .inf file and selecting install. You have to manually specify the .inf file when updating/installing a new driver.

https://www.youtube.com/watch?v=S97hJsg4H20

sergiojavdiaz, 2018-09-16:
Hey bro. I finally made it work, just was a matter of using an older release (Drivers.2358.zip). Try it out and tell me if it worked for you

ziyabnajeeb, 2018-09-16:
Hi there, I just freshly installed windows 10 then bootcamp driver version 5.1, whenever I'm trying to install Drivers.2358.zip, I am getting a code 10 error in device manager...

I have tried to install certificate for current user and for local machine as well, but nothing happens...
Can you explain which store do I have to install the certificate, may b this is what I'm doing wrong

sergiojavdiaz, 2018-09-16:
> Hi there, I just freshly installed windows 10 then bootcamp driver version 5.1, whenever I'm trying to install Drivers.2358.zip, I am getting a code 10 error in device manager...
> I have tried to [...]

No man don't use device manager to install it! Just follow the instructions provided on the main page of this website.

sergiojavdiaz, 2018-09-16:
Go to Drivers>x64>ReleaseSigned>AmtPtpDevice. Right clic on AmtPtpDevice.inf and install it.

Adamsapplespie, 2018-09-17:
Right so I reinstalled it from that .inf file indicated and went through the installation process. Do I need to reboot or something? Because I don't see any difference in my trackpad.

sergiojavdiaz, 2018-09-18:
Bro are you installing the old release? (Drivers.2358.zip)

Adamsapplespie, 2018-09-18:
Nope, I'm installing 2484.

sergiojavdiaz, 2018-09-18:
Well, that's the problem...

sergiojavdiaz, 2018-09-18:
YES!!!

Adamsapplespie, 2018-09-18:
Alright thanks. Ill try that when I get the chance.

closed 2018-09-27 by imbushuo