imbushuo/mac-precision-touchpad #124: Please do an install tutorial for all of us

sergiojavdiaz, 2018-09-15:
Why you keep making you deaf about we all that are crying for help to install the drivers?
How do you expect that people help you or found you if we're not even able to install what you're working on?

Please comprehend that the mayority of us don't know anything about code, we are all newbies please do a video accordingly.

sergiojavdiaz, 2018-09-15:
I'm so mad about the fact that I can't install the drivers

sergiojavdiaz, 2018-09-15:
Further I could go was getting error code 10 on device manager and loosing my cursor. This is so frustrating :(

sergiojavdiaz, 2018-09-16:
I finally made it work, just was a matter of using an older release (Drivers.2358.zip)

jonalbo, 2018-09-20:
How do I get to that older release?

cukabeka, 2019-01-20:
Hi, although I don't agree with the TE's complaints, I would appreciate a lot to have some elaborated instruction on the Installation process:

- When installing the Certificate via right-click + install certificate, I'm asked whether to save it to the local user or the local Computer. Which one is the right choice?
- After that I'm prompted with the question where to save the certificate: automatically or manually. Also here, I'm not sure what makes the Installation work. I Chose the automatic Option as I have no idea what to choose manually.
- Then after I installed AmtPtpDevice.inf with right-click + install from Explorer, I'm not sure what the expected behaviour is afterwards: Can I use gestures right out of the box immediately? Do I Need to restart Windows? Is there a Control Panel of some sort? How do I know the Drivers are active?

I'm asking this as I have no success yet. Maybe These Informations in the install section could help a Little to find out how this should look like.

Anyways, thanks for the amazing work on that!

cukabeka, 2019-01-20:
Haha, I found that on reddit some People searching for a trackpad++ alternative had the same questions: https://www.reddit.com/r/mac/comments/91251n/a_better_and_free_alternative_to_trackpad_for/ :)

sergiojavdiaz, 2019-01-20:
It doesn't matter what you choose, if you choose user or computer, or if you choose automatic or manual. What matters is that you install the right drivers for your machine, I installed the older 2358 drivers in my mid 2013 MacBook Air, and they work like a charm. They should work right away after you right clicked and installed AmtPtpDevice.inf.