imbushuo/mac-precision-touchpad #127: MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports) Device ID

flagersinc, 2018-09-20:
The new model of MacBook Pro routes trackpad and keyboard data through the new embedded T2 chip, this is why the device is listed within the USB Device Tree of the iBridge Bus.
[image]
[image]

FudiHub, 2018-10-17:
What else do you need to make your great driver support the MBP mid 2018 13 inch model? I am really looking forward to it. Thanks! Donation incoming.

soooch, 2018-11-01:
Could we use the existing USB driver for this, or would some relatively drastic changes have to be made on account of the T2?

adriankwil, 2018-11-20:
> Could we use the existing USB driver for this, or would some relatively drastic changes have to be made on account of the T2?

I have tried to install the current release for the USB trackpad but I get the error " device could not start error code 10 ". I expect this is something to do with the T2 and/or my incompetence with installation XD

imbushuo, 2019-07-30:
Supported as of today

closed 2019-07-30 by imbushuo