imbushuo/mac-precision-touchpad #130: High memory usage after 2 days

cubesu9ar, 2018-09-29:
I compiled and installed SPI driver, but after 2 or 3 days( I'm not use this PC too much, maybe 3~4 hours total), system becomes very slow. 

Checked the task manager, memory usage becomes 100%, even I closed all apps.
After change back to old version of SPI touchpad driver, memory usage reduced to 18%

Memory leak?

kazenoai, 2018-10-21:
I think I also encounter this issue. The RAM occupation is almost 100%. It down to normal when I stop touch pad driver.

imbushuo, 2018-11-06:
I will look into it. If you can trigger a manual system crash when the usage is high, it might be helpful to me.

nvella, 2019-07-29:
Confirming this on 2015 13" rMBP. I only restarted earlier today, however I have had a few sleep/wakes since then, and have connected/disconnected various different peripherals and external monitors.

enricmarques, 2019-08-04:
I can confirm the same issue on my MacBook 12" 2017, using latest Windows 10 1903. I usually have my laptop connected to an external monitor, keyboard and mouse, and I've auto-sleep disabled on purpose. During the last 3 days, I've seen this situations:

- Memory usage = 99% (I have 16GB of RAM). Computer was really slow and I had to manually restart it.
- 3 time computer had restarted. After checking Event Viewer, it seems system crashed.

I've reverted to default Apple driver, and both issues are gone. I think both issues are related: I guess system crashed because of a high memory usage.

I'm a developer, so let me know if I can do anything to help you with this issue.

imbushuo, 2019-08-05:
I am currently looking into the matter on SPI driver and aware of one piece of memory allocation that not being recycled (at a hot path). Will spin up a device to verify it shortly.

imbushuo, 2019-08-05:
@enricmarques If you can provide me a memory dump after crash (located in `C:\Windows\MEMORY.DMP`), it will be helpful for me to understand the inner issue.

Please note that this file might contains Personal Identifiable Information (PII), so please contact me using the email provided in my GitHub profile, and avoid handling personal sensitive information during the test session if you need to capture a new dump file.

enricmarques, 2019-08-05:
Hi @imbushuo,

There is no C:\Windows\MEMORY.DMP file. I've checked my "Startup and Recovery" system settings, and my computer should write an "Automatic memory dump" at this location when the system crashes, but it seems the file is not generated. 

I've checked the event viewer, and there is a Event 41 just when the computer restarted. The event description is: "The system has rebooted without cleanly shutting down first. This error could be caused if the system stopped responding, crashed, or lost power unexpectedly". I have checked events of the previous minutes, and there are some errors from services that says something like "There are not enough system resources to complete the operation". I think Windows crashed without generating a memory dump file. Maybe there aren't enough system resources to save the dump.

imbushuo, 2019-08-05:
@enricmarques Never mind, I've narrowed down to the root cause right now.

enricmarques, 2019-08-05:
@imbushuo. Great! I was about to use the NotMyFault tool from SysInternals (https://docs.microsoft.com/en-us/sysinternals/downloads/notmyfault) to manually generate a memory.dmp file.

imbushuo, 2019-08-05:
Root cause: something still takes the reference even though input routine completes, and hence the non paged pool memory for input buffer cannot be freed.

closed 2019-08-05 by imbushuo

imbushuo, 2019-08-05:
Fixed in release 3544.

[image]

> THERE'S ABSOLUTELY NO LEAK - Phil Swift

enricmarques, 2019-08-08:
I can confirm the leak is gone, as I had it installed for two days without crahes or memory issues.

commit 1f1cca45 (2019-08-05): SPI: Fix #130

It turns out that I did a bunch of incredibly stupid things in the summer of 2018. Excessive HID read requests were issued at 10ms interval, which filled up the pool in a few days. In addition, the kernel thread event synchronization was also terribly implemented.

I SAWED EXCESSIVE REQUESTS IN HALF!
...THERE'S ABSOLUTELY NO LEAK!