imbushuo/mac-precision-touchpad #131: Device ID, MacBook Pro 15 inch, Mid 2015, with retina screen

chemPolonium, 2018-09-30:
[image]

cycorey, 2018-11-21:
yes, my mac(MacBook Pro 15 inch, Mid 2015, with retina screen), Device ID is the same.

I installed the usb version, but not see any thing more in the control panel.

If @imbushuo could merge the support for MacBook Pro (15-inch, 2015), It will be a great help to me!

Terry2001, 2019-04-14:
my mac is same. hope for support, thanks very very much~~~

imbushuo, 2019-04-14:
We actually have a few code for Mid 2015 device merged in. I don’t have the device (the original device owner has transferred the device to someone else), so it may be tricky for me to test it at this moment.

Plus I am dealing with intensive academic workload. If there is any update (likely summer), I will let you know.

Terry2001, 2019-04-14:
Thanks for reply soon.  Maybe I can research how to test the driver on my device. I will try it in a few days. Could you tell me where I can find the document about setting a testing or development environment?

imbushuo, 2019-04-14:
Here should have everything you need: https://docs.microsoft.com/en-us/windows-hardware/drivers/

By the way, there is no need for enabling test signing on UMDF drivers (in your case). And you don’t need to set up test agent (that’s for automation test).

For some reason, user-mode debugger never works for this driver, so most time I used WPP tracing and TraceView to help me diagnosing issues: https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/traceview

Terry2001, 2019-04-14:
@imbushuo Thanks for your great help.  After modify some code,  the touchpad is already working well in debug version.  But I‘m not sure the code change is correct.  I will debug it more.

commit link as below, could you review it if you have time.
https://github.com/Terry2001/mac-precision-touchpad/commit/7a941b5daee3ad53feb2a1b890402b200548187b?diff=split

imbushuo, 2019-04-14:
Hi,

This looks good to me. Could you kindly send me a PR?

Thanks,
Bingxing Wang

Terry2001@7a941b5?diff=split<https://github.com/Terry2001/mac-precision-touchpad/commit/7a941b5daee3ad53feb2a1b890402b200548187b?diff=split>

Terry2001, 2019-04-16:
Let me test it a week and check some documents to confirm the modification is right.  Then I will send a PR.

imbushuo, 2019-04-25:
Merged through https://github.com/imbushuo/mac-precision-touchpad/pull/179.

closed 2019-04-25 by imbushuo