imbushuo/mac-precision-touchpad #134: The cursor will go to the screen bottom when I touch the panel top position. Macbook 2015 13.3

kazenoai, 2018-10-01:
The cursor will to to the screen's bottom immediately when I figure touch the top position of touch panel.
Computer: Macbook 2015 13.3
OS: Windows 10, 1803
Driver type: SPI driver(07/14/2018,14.56.52.89)

kazenoai, 2018-10-02:
[image]
I change the code of input.c, and try it in my computer. The cursor does not lost and appear at screen's bottom area when I touch the top area of touch panel. And this issue also will occur when I touch the touch panel left side, and the cursor will appear the right side edge of screen.

RLst, 2018-10-08:
Yes I think this happens to me too. I have a MacBook Pro retina late 2013. When using it would seem to randomly teleport to the bottom or around bottom right corner of the screen and then ends up opening some other program. Extremely annoying

kazenoai, 2018-10-11:
@RLst You can find the position of that code(SPI: input.c, USB: InputInterrupt.c), and try to modify the code and run the driver on your computer.

imbushuo, 2018-11-29:
@kazenoai Can you send a PR to me?

kazenoai, 2018-11-30:
I have sent  PR to you! Your project is great. Thank you.