imbushuo/mac-precision-touchpad #138: Device ID for MacBook Pro 2017 15" touch bar

Dvornik, 2018-10-03:
Model Name:	MacBook Pro
  Model Identifier:	MacBookPro14,3  

SPI:
 Product ID:	0x0278
 Vendor ID:	0x05ac (Apple Inc.)

Confusingboat, 2018-10-12:
10/10 will donate if you get this one working.

Sotam, 2018-11-09:
It's already working for your device. I have the same machine and installed the SPI driver, which can be found [here](https://github.com/imbushuo/mac-precision-touchpad/issues/109).

themikesam, 2018-12-03:
not working for MacBook Pro 2017 15" with touch bar.

themikesam, 2018-12-10:
trackpad++

Sotam, 2019-01-08:
I used the file from this link: https://github.com/imbushuo/mac-precision-touchpad/issues/109#issuecomment-409779286 and disabled driver signature enforcement. After installing I re-enabled it. Not sure why it's not working for you, I have the same model.

imbushuo, 2019-10-21:
0278 should be supported per https://github.com/imbushuo/mac-precision-touchpad/blob/702e8c518ab052272c2f5f62c33d12a76eb160e2/src/AmtPtpDeviceSpiKm/AmtPtpDeviceSpiKm.inf.

closed 2019-10-21 by imbushuo