imbushuo/mac-precision-touchpad #143: My MacBook Air 2017(13-inch) cannot activate the driver. (Attached my install steps)

primelyw, 2018-10-19:
I tried to install your driver according to your instructions. After installing it, I cannot move the cursor using my internal touchpad.
     Here is my installing steps.
     1) Download Drivers.2358.zip.
     2) Follow your instructions in "README.md"
     3) Update my driver in Windows Device Manager

     Please Help,Thanks!
     
[image]

[image]

mjw3, 2018-10-19:
You must enable test signing for the drivers at this time.
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option

imbushuo, 2019-10-21:
Supported added with https://github.com/imbushuo/mac-precision-touchpad/commit/304062d4d2c2d732115164da87d35931d2d4d756, and will be rolled out with next release. Note that this is a fallback support, feel free to open a ticket if the experience is not optimal.

closed 2019-10-21 by imbushuo