imbushuo/mac-precision-touchpad #147: Macbook Pro 2018 15 Touchbar 15,1

naknak237, 2018-11-06:
[image]
[image]

deg0nz, 2018-11-08:
Almost identical with 2018 MacBook Pro 13'' with touch bar:

MacBookPro15,2

  Product ID:	0x027b
  Vendor ID:	0x05ac (Apple Inc.)

stanczyk4, 2018-11-10:
[image]

Macbook 2018 15" w/ Touch Bar. I have Windows boot camped if you need to remote in for testing. Would love to have real touchpad support to do windows development on my Mac!

as-com, 2018-11-22:
[image]
Here's my 15" 2018 MacBook Pro

FZHBW, 2019-06-29:
[image]

imbushuo, 2019-07-30:
Supported as of today

closed 2019-07-30 by imbushuo