imbushuo/mac-precision-touchpad #149: Macbook Pro 2016 15 Touchbar MacBookPro13,3

zhangwujun666, 2018-11-28:
[image]
[image]

LukasMendez, 2018-12-11:
Did you ever find out if it was possible to use this? I have the exact same machine

zhangwujun666, 2018-12-16:
Not really, because the installation of SIP has not released yet. And I try to use the developer model to run but failed.

imbushuo, 2019-10-21:
0278 should be support in https://github.com/imbushuo/mac-precision-touchpad/blob/702e8c518ab052272c2f5f62c33d12a76eb160e2/src/AmtPtpDeviceSpiKm/AmtPtpDeviceSpiKm.inf.

closed 2019-10-21 by imbushuo