imbushuo/mac-precision-touchpad #166: How to adjust touch and press sensitivity?

markreg, 2019-01-29:
I'm using the MacBookPro11,5 (trackpad with magnets) and have installed this driver.

How can I adjust the sensitivity of touch and button press? Is there any code I can change to adjust it?

Thanks

PieroAlvarez, 2019-03-14:
I have the same request for my Magic Trackpad 2.

The Magic Trackpad 2 works correctly but the amount of surface of the finger needed to activate the movement of the cursor is too large so I can not move it with the tips of my fingers (with the Apple drivers is possible).

freecodecampster, 2019-05-28:
Try 1802.M2 Interim Release. Go to settings in Windows-Devices-Touchpad to change settings.

imbushuo, 2019-10-21:
Unfortunately there's no way to adjust the sensitivity at this moment. You can try adjust the sensitivity in Windows settings but I don't think that will be really helpful. I might expose an interface to adjust the sensitivity curve in the future.

closed 2019-10-21 by imbushuo