imbushuo/mac-precision-touchpad #169: MacBook Air Late 2018 - Not working

Fleischhauer, 2019-02-16:
Uhm, I would like to post the device-ID but in the system information it says there is no SPI-Device o.O

Could you tell me how I can get it?

I would like to get this to work.

Fleischhauer, 2019-02-16:
On Windows there is a lot in the device manager:
https://i.imgur.com/3ZaPcAZ.png

The top one has this ID:
https://i.imgur.com/FB51slp.png

Is that what you need?

borb, 2019-02-20:
That USB ID may be the keyboard; whilst the IDs may be similar, and I'm not discounting the possibility of it actually being the touchpad, is there another ID lacking a driver?

Fleischhauer, 2019-02-20:
Nothing lacks a driver.

I think the multi touch thingy is the trackpad.
It uses this driver atm: applemtp64.inf_amd64_82b70b68f5f7b0f9 für Geräteinstanz-ID USB\VID_05AC&PID_027A&MI_02\7&190263F6&0&0002
mtp = multi track pad maybe?

Fleischhauer, 2019-02-20:
I tried two drivers now with that, and the trackpad stopped working entirely lol. At least it seems to be it.
Now I just have to find the right one...

Fleischhauer, 2019-02-20:
Tried everyone in releases. None works.

imbushuo, 2019-07-30:
Supported as of today

closed 2019-07-30 by imbushuo