imbushuo/mac-precision-touchpad #171: SPI trackpads

knittycatkitty, 2019-02-19:
For anyone who is looking to install this driver on SPI devices such as Macbook12,1 and up

The proper drivers are on his site
https://ptp-release.imbushuo.net/Apple%20PTP%20Trackpad%20(SPI%20Kernel%20Mode)/fc1ebcf0dffc1f921d70a10c461766eb54ccd38b/AmtPtpDeviceSpiKm/x64/Release/

It is not signed so you will need to start Windows 10 with Disabled Driver Signature Enforcement to install it.

It works beautifully, I recommend trying it on 2016 and up, as far as I can tell from the INF file, SPI devices are all the same from 2015 and up mostly.

SPI could most likely be synonymous with Force Touch trackpads.

charlespick, 2019-02-19:
Tried on my MBP 13,1. Disabled driver signature enforcement, installed driver in device manager. Cursor disappears. Restart doesn't fix it. Any suggestions?

knittycatkitty, 2019-02-20:
> Tried on my MBP 13,1. Disabled driver signature enforcement, installed driver in device manager. Cursor disappears. Restart doesn't fix it. Any suggestions?

Try a different version, had that issue as well

https://ptp-release.imbushuo.net/Apple%20PTP%20Trackpad%20(SPI%20Kernel%20Mode)/

Fleischhauer, 2019-02-20:
I tried that one on Macbook Air 2018, didn't work :/

Fleischhauer, 2019-02-20:
I try some others now. And I get code 10 now, device couldn't be startet.
Also, why is there a folder with another .inf? Do I need that? Do I use that inf or the one in the root folder?

knittycatkitty, 2019-02-20:
> I try some others now. And I get code 10 now, device couldn't be startet.
> Also, why is there a folder with another .inf? Do I need that? Do I use that inf or the one in the root folder?

What is your device ID? I think for Macbook Air 2018 it was released a few months after the author's last commit on this project.

First step is to put the Apple SPI Touchpad device back to the original drivers.

Also personally I have not seen a difference, I notice all driver packages for Windows come like that though. Not sure why

Fleischhauer, 2019-02-20:
HID are:
USB\VID_05AC&PID_027A&REV_0913&MI_02
and
USB\VID_05AC&PID_027A&MI_02

Is that right?

knittycatkitty, 2019-02-20:
> HID are:
> USB\VID_05AC&PID_027A&REV_0913&MI_02
> and
> USB\VID_05AC&PID_027A&MI_02
> 
> Is that right?

Yep let me see what I can do

knittycatkitty, 2019-02-20:
> HID are:
> USB\VID_05AC&PID_027A&REV_0913&MI_02
> and
> USB\VID_05AC&PID_027A&MI_02
> 
> Is that right?

Can you send me the SPI ones?

Fleischhauer, 2019-02-20:
Where can I find them? What is SPI? ^^'

knittycatkitty, 2019-02-20:
> Where can I find them? What is SPI? ^^'

Device Manager -> Human Interface Devices

[image]
Ignore the caption

Fleischhauer, 2019-02-20:
I don't have a SPI one there.
It looks like this: https://i.imgur.com/3ZaPcAZ.png

Fleischhauer, 2019-02-20:
Maybe the new trackpad is not spi anymore or something?

Fleischhauer, 2019-02-20:
The second one is the one that makes my mouse disappear when I deactivate it.

knittycatkitty, 2019-02-20:
> The second one is the one that makes my mouse disappear when I deactivate it.

I see, can you send me the VID for those? Hard to tell if they are USB or SPI

Fleischhauer, 2019-02-20:
VID, is that not what I send you?
USB\VID_05AC&PID_027A&REV_0913&MI_02
and
USB\VID_05AC&PID_027A&MI_02

knittycatkitty, 2019-02-20:
> VID, is that not what I send you?
> USB\VID_05AC&PID_027A&REV_0913&MI_02
> and
> USB\VID_05AC&PID_027A&MI_02

Try this.
https://drive.google.com/open?id=1HK-hbGq4Jd0ns4WnMvXyPoRDtP-R5TMi

Fleischhauer, 2019-02-20:
I only need that one file?

Fleischhauer, 2019-02-20:
With that I get a new error: "This is not for x64 based systems"

knittycatkitty, 2019-02-20:
> With that I get a new error: "This is not for x64 based systems"

https://drive.google.com/open?id=1iakWSv1KgYYI145g0zzLhG5zf1wsNENi

Fleischhauer, 2019-02-20:
The same error TT

knittycatkitty, 2019-02-20:
> The same error TT

Last one haha if it doesn't work.. ahh I can't help! 

https://drive.google.com/open?id=1fcsU8V8vSHIE27vWsq31W7f6pSU53T1m

Fleischhauer, 2019-02-20:
Still the same.
Thanks for trying!

Fleischhauer, 2019-02-20:
Tried everyone now. None works.

ucristian3, 2019-02-25:
thanks for the explanation!
I used the same method 3 months ago and i'm using the this driver since that day.
everything work as is should the only big problem I encouter is that I get BSOD when oppening the lid after some time.
Macbook pro (2017) not touchbar 14,1

mattbeau, 2019-03-23:
> thanks for the explanation!
> I used the same method 3 months ago and i'm using the this driver since that day.
> everything work as is should the only big problem I encouter is that I get BSOD when oppening the lid after some time.
> Macbook pro (2017) not touchbar 14,1

so which one did you install? I tried all of them and keep getting error 10 on my MacBook Pro 2017 nTB 14,1

ucristian3, 2019-03-23:
> so which one did you install? I tried all of them and keep getting error 10 on my MacBook Pro 2017 nTB 14,1

I used this one here....
dont forget to enable TEST MODE     bcdedit -set TESTSIGNING ON    in cmd and and restart PC,after this you can proceed to installation proces

[Macbook.Pro.2016.WinPrecisionTouchpad.zip](https://github.com/imbushuo/mac-precision-touchpad/files/2999674/Macbook.Pro.2016.WinPrecisionTouchpad.zip)

mattbeau, 2019-03-24:
> thanks for the explanation!
> I used the same method 3 months ago and i'm using the this driver since that day.
> everything work as is should the only big problem I encouter is that I get BSOD when [...]

Awesome, unfortunately it was an empty zip file but I googled the file name and found it. here it is for others wanting to try it out too :)

[Macbook.Pro.2016.WinPrecisionTouchpad (1).zip](https://github.com/imbushuo/mac-precision-touchpad/files/2999966/Macbook.Pro.2016.WinPrecisionTouchpad.1.zip)

thekip, 2019-04-17:
> Awesome, unfortunately it was an empty zip file but I googled the file name and found it. here it is for others wanting to try it out too :)
> 
> Macbook.Pro.2016.WinPrecisionTouchpad (1).zip

Hooray, exactly this version works except it always crash the system when you close the lid.

(all other versions get error 10)

bpwned, 2019-04-19:
> Hooray, exactly this version works except it always crash the system when you close the lid.
> 
> (all other versions get error 10)

I can confirm this on my MacbookPro13,3.

imbushuo, 2019-07-21:
As of today, SPI trackpad does not require TestSigning. See updated README.

closed 2019-07-21 by imbushuo

carrottopadc, 2019-11-22:
Hello! If anyone could help me, that would be fantastic! I used the 2016 mbp drivers for my 2019 15inch mbp and it worked great. I returned that to apple and purchased the 16 inch macbook pro, and not a single one of the drivers work! Spent nearly 4 hours pulling my hair out over this-Any help would be greatly appreciated.
Thanks in advance all,
Andy C