imbushuo/mac-precision-touchpad #172: MacBookPro 15,3 (Vega 16/20) USB Device ID's

goldmastersims, 2019-03-01:
[image]
[image]
[image]
[image]
[image]

goldmastersims, 2019-03-01:
Nothing in the SPI devices

imbushuo, 2019-07-21:
027c is supported in https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3529, please try and provide feedback.

closed 2019-07-21 by imbushuo