imbushuo/mac-precision-touchpad #174: MacBook Pro Retina 13" Early 2015 Support

mau-lopez, 2019-03-04:
Hi imbushuo, this is the info:

  Product ID:	0x0273
  Vendor ID:	0x05ac (Apple Inc.)
  ST Version:	6.22
  MT Version:	4.10

MacBook Pro Retina 13" Early 2015

Thanks.

mau-lopez, 2019-03-04:
By the way, if I can be of any help in testing the support for this model, please let me know!

nvella, 2019-03-18:
I have gotten the driver to work on my Early 2015 rMBP by compiling the latest source myself with VS and WDK. Only issue is that I have to enable driver signing test mode on Windows or else the driver will not load, as VS signs with a test certificate. Personally I don't wish to leave test mode enabled as it's a considerable security risk. Will there be a new signed release soon @imbushuo?

poomonnateluercha, 2019-03-22:
I confirm @nvella that this does work on early 2015 13" rMBP by compiling the latest source, with driver signing test mode on

qwqVictor, 2019-04-25:
I'd like to test it on my MacBookPro12,1. But the WDK is too big for me to download and install. So could any one send me a copy of the driver release? I'll be very grateful for this.

imbushuo, 2019-04-25:
@qwqVictor I've fixed CI, so if you can check the link on README.md file, you should be able get a copy of signed driver.

imbushuo, 2019-04-25:
I've merged it through https://github.com/imbushuo/mac-precision-touchpad/pull/179. So right now I will go ahead and close the issue.

closed 2019-04-25 by imbushuo

fearthewopr, 2019-06-07:
Hi! I tried to access the CI builds, but don't see the one that matches the commit ID for the addition of the 2015 MBP that Terry was involved with.  Am I just totally missing something here?

combdn, 2019-06-28:
@fearthewopr I just went through all of the builds and checked for April 2019 (the month Terry's commit was added) 😄 
Here is [the one you need](https://ptp-release.imbushuo.net/Apple%20PTP%20Trackpad%20Build/57075ea9bc0d9006d6e4e5e50098e4bef1e15a4f/) (works for me).